openscad/openscad #3155: Less intrusive autocomplete

thehans, 2019-12-10:
The autocomplete editor feature is nice, but I have been getting some false-positive matches recently, which I'm finding tedious to avoid.

For example I needed a function that returns mathematical modulo.  (like `%` operator but always return non-negative values).  So I wrote a function called `mod(x,y)`, but then whenever I type to call the function: `mod(` it automcompletes to `module(`
Also, trying to call a function named `f()` autocompletes to `false(`, or a function named `rot(...`, autocompletes to `rotate(`, etc.

Tab or Enter keys (and maybe others?) already are ways to accept the autocomplete, so I'd rather not have `(` trigger it too.

`module(` isn't even correct syntax anyways (without a module name).

Secondly, I think autocomplete should be disabled when typing inside comments.

MichaelAtOz, 2019-12-11:
Also, I think the pop-up dialogue should be moved, say, ~2 lines away from the cursor and/or to the right. Frequently you want to see the adjacent lines. Particularly a problem at the bottom of the page:
[image]

t-paul, 2019-12-11:
I agree that it would be a useful change, but the implementation of those features is inside QScintilla, so the control on our side is limited.
I found that setting the trigger to 2 characters helps a lot with the misfires (but would not help for `mod`). Maybe there's at least some control over the keyboard handling as I think the most problematic part is the auto-accept of the selected value when just typing.

Sharma-Hrishabh, 2020-02-01:
Hey @t-paul, I want to work on this issue. Can you guide me, from where should I begin?

rcolyer, 2020-02-02:
Hi @Sharma-Hrishabh, if memory serves these features came in with the merge of PR #2975.  That should point you to the relevant subset of the source code.

t-paul, 2020-02-16:
Fixed via  #3207.

closed 2020-02-16 by t-paul

diabl0w, 2020-08-13:
how do I enable auto-complete? I cannot find documentation on this anywhere and it is not in preferences from what I can see.

Sharma-Hrishabh, 2020-08-13:
hey @diabl0w , the option to enable/disable autocomplete is present in Preferences->Editor. [ at the bottom ]

diabl0w, 2020-08-13:
> hey @diabl0w , the option to enable/disable autocomplete is present in Preferences->Editor. [ at the bottom ]

Thanks for the reply, is there a certain version that is required? The option is not there for me. Under Help-> About, it says I am running version 2019.05 
 
This is what my Preferences -> Editor looks like:

[image]

MichaelAtOz, 2020-08-13:
[Development Snapshot](http://www.openscad.org/downloads.html#snapshots)