openscad/openscad #3157: Cache issue: previous and current geometry both rendered with F6 after changing the scad

tturpin, 2019-12-12:
The following scad code (which I simplified as much as I could) produces an incorrect rendering with F6 when y is changed from 4 to -4:
```
y=4;

module bottom_ring() {
     translate([0,y]) cube(4);
     translate([-5,0]) cylinder(r=2,h=2);
}

bottom_ring();

difference() {
     translate([5,0]) cube(3);
     cube(2);
}
```
Reproduction steps:
- open the file in OpenScad (tested with master 24aa70f5284bad94b06f6b3611114a8b2d14f3a4)
- render (F6)
- change the value of y from 4 to -4
- render again

The cube of size 4 is rendered at both positions. Flushing caches fixes the issue.

MichaelAtOz, 2019-12-12:
Does not happen for me with 2019.05 or 2019.11.20 (latest snapshot) on Windows or 2019.11.24.nightly on Debian.

thehans, 2019-12-12:
I was able to reproduce on Linux nightly, but it only happens when lazy union feature is *disabled*.

MichaelAtOz, 2019-12-12:
So what does lazy enabled mean? Enabled means automatic union, ie old style, it does it for you?

MichaelAtOz, 2019-12-12:
OK, I see, lazy ie it doesn't do extra work.
I also reproduced this issue with lazy union disabled with 2019.11.20 on W7 & 2019.11.24 on Debian.
Also another display oddity, I'll report in the Lazy union issue.

thehans, 2019-12-16:
I think this is now fixed with PR #3159

thehans, 2019-12-16:
Fixed in nightly

closed 2019-12-16 by thehans