openscad/openscad #3158: Preview and Render results completely different (nested for loops)

speed-3d, 2019-12-15:
When Previewing the following code, everything looks correct.  When I Render, I get completely different and unexpected results.  Am I doing something wrong, or is this an openSCAD bug?

[image]
[image]

*********** start code *********
for (x = [0:1]) {
    for (y = [0:1]) {
        translate ([(x*50),0,(y*30)]) {
            difference () {
                cube ([55,35,35]);
                translate ([5,-0.01,5]) {
                    cube ([45,30,25]);
                }
            }
            difference () {
                translate ([27.5,15,7]) {
                    sphere (d=29, $fa=1);
                }
                translate ([0,0,-20]) {
                    cube ([55,35,20]);
                }
            }
        }
    }
}
***********  end code *********

OpenSCAD 2019.11.20
Ubuntu 18.04.3 LTS

OpenSCAD Version: 2019.11.20
System information: Linux 5.0.0-37-generic #40~18.04.1-Ubuntu SMP Thu Nov 14 12:06:39 UTC 2019 x86_64 Ubuntu Core 16 4 CPUs 7.69 GB RAM
User Agent: OpenSCAD/2019.11.20 (Linux x86_64; Ubuntu Core 16)
Compiler: GCC "5.4.0 20160609" 64bit
MinGW build: No
Debug build: No
Boost version: 1_58
Eigen version: 3.2.92
CGAL version, kernels: 4.14.1, Cartesian, Extended_cartesian, Epeck
OpenCSG version: OpenCSG 1.4.2
Qt version: 5.5.1
QScintilla version: 2.9.1
InputDrivers: 
GLib version: 2.48.2
lodepng version: 20180910
libzip version: 1.0.1
fontconfig version: 2.11.94
freetype version: 2.6.1
harfbuzz version: 1.0.1
lib3mf version: (not enabled)
Application Path: /snap/openscad-nightly/1489/usr/bin
Documents Path: /home/speed/snap/openscad-nightly/1489/.local/share
User Documents Path: /home/speed/snap/openscad-nightly/1489
Resource Path: /snap/openscad-nightly/1489/usr/share/openscad-nightly
User Library Path: /home/speed/snap/openscad-nightly/1489/.local/share/OpenSCAD/libraries
User Config Path: /home/speed/snap/openscad-nightly/1489/.config/OpenSCAD
Backup Path: /home/speed/snap/openscad-nightly/1489/.local/share/OpenSCAD/backups
OPENSCADPATH: 
OpenSCAD library path:
/home/speed/snap/openscad-nightly/1489/.local/share/OpenSCAD/libraries
/snap/openscad-nightly/1489/usr/share/openscad-nightly/libraries

OPENSCAD_FONT_PATH: 
OpenSCAD font path:
/usr/share/fonts
/usr/local/share/fonts
/home/speed/snap/openscad-nightly/1489/.local/share/fonts
/home/speed/snap/openscad-nightly/1489/.fonts
/snap/openscad-nightly/1489/usr/share/fonts
[... 294 lines cut ...]
GL_KHR_no_error
GL_KHR_robust_buffer_access_behavior
GL_KHR_robustness
GL_MESA_pack_invert
GL_MESA_shader_integer_functions
GL_MESA_texture_signed_rgba
GL_MESA_window_pos
GL_NV_blend_square
GL_NV_conditional_render
GL_NV_depth_clamp
GL_NV_light_max_exponent
GL_NV_packed_depth_stencil
GL_NV_primitive_restart
GL_NV_texgen_reflection
GL_NV_texture_barrier
GL_NV_texture_env_combine4
GL_NV_texture_rectangle
GL_OES_EGL_image
GL_OES_read_format
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_border_clamp
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SUN_multi_draw_arrays

MichaelAtOz, 2019-12-15:
Looks like a bug, reproduced with 2019.11.20, it works properly with the release version 2019.05.

 

  _____  

(#3158)

 

 
<https://user-images.githubusercontent.com/57547534/70871289-69794500-1f63-11ea-9b7d-6587ed00c4b9.p
ng> openSCAD-preview
 
<https://user-images.githubusercontent.com/57547534/70871290-69794500-1f63-11ea-91ea-746cb1c1474f.p
ng> openSCAD-render

<https://github.com/openscad/openscad/issues/3158?email_source=notifications&email_token=AAPBVYM2P5
AKRZ3RPR72WZ3QY27JTA5CNFSM4J3CVK6KYY3PNVWWK3TUL52HS4DFUVEXG43VMWVGG33NNVSW45C7NFSM4IATERKA>  it on
<https://github.com/notifications/beacon/AAPBVYJVL7BEWVH4BZ4EZZ3QY27JTA5CNFSM4J3CVK6KYY3PNVWWK3TUL5
2HS4DFUVEXG43VMWVGG33NNVSW45C7NFSM4IATERKA.gif>

thehans, 2019-12-16:
@speed-3d FYI to post code snippets on github, wrap them in 3 back-tick characters ``` on a separate line before and after your code.

Otherwise the code is interpreted as markdown and some character like multiplication `*` get interpreted as italics, whitespace indentation doesn't display correctly, etc. 
So the code is no longer easy to copy-paste into OpenSCAD, 
Or to put a single line of code inline with text, you can wrap in single backticks: `example`

Anyways, I've already edited your post with to fix the code display, just mentioning for future reference.

rcolyer, 2019-12-16:
Simplified fast-rendering testing code with equivalent bug:

```
for (x = [0:1]) {
  for (y = [0:1]) {
    translate ([(x*50),0,(y*30)]) {
      difference () {
        cube ([55,35,35]);
        translate ([5,-0.01,5]) {
          cube ([45,30,25]);
        }
      }
      translate ([27.5,15,7]) {
        sphere (d=10);
      }
    }
  }
}
```

Note the caching is not behaving well for this either.  Making changes and rendering without flushing caches was inventing some additional anomalous results.

speed-3d, 2019-12-16:
CSG Tree looks normal

```
group() {
  group() {
    multmatrix([[1, 0, 0, 0], [0, 1, 0, 0], [0, 0, 1, 0], [0, 0, 0, 1]]) {
      difference() {
        cube(size = [55, 35, 35], center = false);
        multmatrix([[1, 0, 0, 5], [0, 1, 0, -0.01], [0, 0, 1, 5], [0, 0, 0, 1]]) {
          cube(size = [45, 30, 25], center = false);
        }
      }
      difference() {
        multmatrix([[1, 0, 0, 27.5], [0, 1, 0, 15], [0, 0, 1, 7], [0, 0, 0, 1]]) {
          sphere($fn = 0, $fa = 1, $fs = 2, r = 14.5);
        }
        multmatrix([[1, 0, 0, 0], [0, 1, 0, 0], [0, 0, 1, -20], [0, 0, 0, 1]]) {
          cube(size = [55, 35, 20], center = false);
        }
      }
    }
    multmatrix([[1, 0, 0, 0], [0, 1, 0, 0], [0, 0, 1, 30], [0, 0, 0, 1]]) {
      difference() {
        cube(size = [55, 35, 35], center = false);
        multmatrix([[1, 0, 0, 5], [0, 1, 0, -0.01], [0, 0, 1, 5], [0, 0, 0, 1]]) {
          cube(size = [45, 30, 25], center = false);
        }
      }
      difference() {
        multmatrix([[1, 0, 0, 27.5], [0, 1, 0, 15], [0, 0, 1, 7], [0, 0, 0, 1]]) {
          sphere($fn = 0, $fa = 1, $fs = 2, r = 14.5);
        }
        multmatrix([[1, 0, 0, 0], [0, 1, 0, 0], [0, 0, 1, -20], [0, 0, 0, 1]]) {
          cube(size = [55, 35, 20], center = false);
        }
      }
    }
  }
  group() {
    multmatrix([[1, 0, 0, 50], [0, 1, 0, 0], [0, 0, 1, 0], [0, 0, 0, 1]]) {
      difference() {
        cube(size = [55, 35, 35], center = false);
        multmatrix([[1, 0, 0, 5], [0, 1, 0, -0.01], [0, 0, 1, 5], [0, 0, 0, 1]]) {
          cube(size = [45, 30, 25], center = false);
        }
      }
      difference() {
        multmatrix([[1, 0, 0, 27.5], [0, 1, 0, 15], [0, 0, 1, 7], [0, 0, 0, 1]]) {
          sphere($fn = 0, $fa = 1, $fs = 2, r = 14.5);
        }
        multmatrix([[1, 0, 0, 0], [0, 1, 0, 0], [0, 0, 1, -20], [0, 0, 0, 1]]) {
          cube(size = [55, 35, 20], center = false);
        }
      }
    }
    multmatrix([[1, 0, 0, 50], [0, 1, 0, 0], [0, 0, 1, 30], [0, 0, 0, 1]]) {
      difference() {
        cube(size = [55, 35, 35], center = false);
        multmatrix([[1, 0, 0, 5], [0, 1, 0, -0.01], [0, 0, 1, 5], [0, 0, 0, 1]]) {
          cube(size = [45, 30, 25], center = false);
        }
      }
      difference() {
        multmatrix([[1, 0, 0, 27.5], [0, 1, 0, 15], [0, 0, 1, 7], [0, 0, 0, 1]]) {
          sphere($fn = 0, $fa = 1, $fs = 2, r = 14.5);
        }
        multmatrix([[1, 0, 0, 0], [0, 1, 0, 0], [0, 0, 1, -20], [0, 0, 0, 1]]) {
          cube(size = [55, 35, 20], center = false);
        }
      }
    }
  }
}
```

thehans, 2019-12-16:
@speed-3d Thanks for the report.  I've found the source of the error and just waiting for some CI tests to finish running before the fix is merged.

thehans, 2019-12-16:
fixed

closed 2019-12-16 by thehans