openscad/openscad #3162: Windows Installer not functioning - 3 files missing

sproggit, 2019-12-21:
Downloaded OpenSCAD-2019.05-x86-64-Installer.exe... Virus Checks OK, checksum OK. 

Attempted to install making one change - to alter the installation directory to D:\Program Files (x86)\OpenSCAD\. Installation completes successfully with no issues reported, but with the program not becoming visible in the Windows Menu. (Windows 10 Pro, 64-Bit). 

Located the installation directory and created a shortcut for the main binary on the desktop. Double-clicking the shortcut results in 3 error messages, in sequence, reporting files missing:-

EVR.dll
MF.dll
MFPlat.dll

Download the zip file version of the program, expand and explore this package - the above named 3 files do not exist within it. 

Unable to progress beyond this point. 

Repeatable error, even with fresh downloads.

MichaelAtOz, 2019-12-21:
I believe `C:\Program Files\` is 'special' protected, installing elsewhere is not normally supported.
Also ...2019.95-x86-64 is 64bit, `D:\Program Files(x86)` is 32bit not 64bit.
Perhaps try installing in the expected spot?

sproggit, 2019-12-21:
Hi Michael, thank you for the prompt response. 

1. I tried installing using your recommended installation location and I did see a small variation in result. I now get the OpenSCAD program name and icon appearing in the "Start Menu" folder, but when I attempt to execute the program I get the same 3 errors reporting the same 3 files missing... 

2. I tried re-running the installer forcibly using "Administrator Mode" (right click installer binary and select "Run as Administrator") and this produces the same result as above (i.e. program now appears in Start Menu but same 3 dll files reported missing. 

The thing is, as my original bug report noted, when I downloaded the zip file and explored the content, I found no trace of those files, so unless the program expects to find them pre-installed (i.e. from Windows 10) or unless the installer pulls them in dynamically across the network, is it possible that the files are actually missing? 

Happy to experiment further. 

Also, for what it's worth, I have OpenSCAD running perfectly under Linux Mint; the package is integrated within Mint's Software Manager (I'm gessing it is included either by the Debian root build or by Ubuntu, on which Mint is in part based... In other words, I have a fully working OpenSCAD, so this is not a high priority issue - it is just that it would be helpful to be able to use he program without having to reboot into Mint. 

Thank you

nophead, 2019-12-21:
On my Win7 machine all those DLLs are in C:/windows/System32, so I think
they are system files, not part of OpenSCAD.

t-paul, 2019-12-21:
https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/FAQ#OpenSCAD_fails_to_run_refering_to_EVR.dll,_MF.dll_or_MFPlat.dll_on_Windows

closed 2019-12-21 by t-paul