openscad/openscad #3165: Comparisons across types and for function literals

thehans, 2019-12-22:
I am working on a library that provides some general algorithms and utilities to take advantage of function literals, and while testing some features I realized some issues with the builtin comparison operators.  Thse are mostly a problem for example if trying to sort/binary_search a vector (which might become "poisoned" by an `undef`, or by other mixing of types).

A summary of some of the observed issues:
1. Comparisons between bool and number is inconsistent
    ```
    echo(false < 0);  // false
    echo(false > 0);  // false
    echo(false != 0); // true   
    echo(false == 0); // false
    echo(false <= 0); // true
    echo(false >= 0); // true
    ```
    and similar issue between `true` and `1`
2. Comparisons among vectors always return false
3. Comparison among ranges always return false
4. Comparison among function literals always return false.
5. Comparison between differing types always returns false.

I created PR #3164 which addresses points 1-3 (which I think should be the least-controversial).
1. `false` and `true` are now consistently equivalent to `0` and `1`  (same as C++, Python, Javascript, etc.)
2. vectors are now comparable using lexicographical order, via the default `std::vector` operators (but mixed types among elements will still present a problem).
3. ranges are also compared in a way that should be consistent with lexicographical order of the lists which they would expand to.  (Basically the start value is first compared.  If equal, then step value is compared.  If equal, then end value is compared)

For the remaining point 4, I am wondering if OpenSCAD should support comparisons between any two differing types.   
One possible point of reference is to look at how Python handles things since it has many similar qualities such as lists allowing mixed types, `None` analogous to `undef`, function literals, etc. 

I ported our `operators-tests.scad` to Python and posted a [comparison of results here](https://gist.github.com/thehans/a6e15c10fbd24da08754a738a2a42cf5/revisions).  The first "revision" is the Python results (in red), while the second is OpenSCAD (in green).
The Python output was search & replaced in a text editor to match with OpenSCAD output format more closely (see comments in the `operators-tests.py` within the same gist link).
Also the OpenSCAD results here are *before* the fixes from #3164

I tested this on Python 2.7, but one thing to note is that while Python 2 allowed comparison between any type of object, Python 3 seems to have dropped this feature.  It mostly just throws `TypeError`s when attempted.  
I've been told that the python 2 ordering between different types is (mostly) based on lexicographical ordering of the type name.
So the results seem to show an ordering like: `None < number(aka int or float) < list < string`
`None` (or `undef` for us) compares less than any other value, which I think makes sense.

If we did similar to Python 2 then we could basically sort any list regardless of contents.
Another option would be to follow Python 3's lead and throw an error when incompatible types are compared.
If we leave things as they are, a library could still theoretically implement its own ordering like that with a complicated comparison function, but without that it fails *silently* when a sort is attempted on mixed-type elements, leading to undefined behavior.

Lastly there is still the question of if we can compare function literals in any meaningful way. 
Currently a function literal is not even equivalent to itself, which I think is problematic.

One idea is that the `toString()` of functions could be compared.  This could at least give a reasonable way to order things, however functions whose definition strings match I think could still return different results depending on their closure context.  So maybe equality specifically could be defined based on the actual pointer address or source location of the function declaration.

kintel, 2020-01-04:
@thehans After merging #3164, could the above text be cleaned up to capture the remaining issues?

thehans, 2020-01-04:
OK, edited

thehans, 2020-08-08:
Resolved by #3383

1) Decided that ordering between different types should be "undefined", and cause for warning.  
2) Function literals can now be checked for equality (basically true if they point to the same function literal object), but ordering of functions is undefined.

See PR comments for more details.

closed 2020-08-08 by thehans