openscad/openscad #3166: Renaming of difference does not work

espr14, 2019-12-23:
version 2019.12.21.ci4161 win64

This does not work.
```
d(){
	cube([100,100,100]);
	cube([10,10,100]);
}

module d(){
	difference() children();
}
```

t-paul, 2019-12-23:
Yes, that's not how it can work. There is an ongoing discussion about #350, specifically #3143 but it's probably still not going to change.

This works:
```
module d() {
	difference() {
        children(0);
        children([1:$children-1]);
    }
}
```

closed 2019-12-23 by t-paul

espr14, 2019-12-25:
This is more safe:

```
module d(){
	if($children==1) children(0);
	if($children>1) difference(){
		children(0);
		children([1:$children-1]);
	}
}
```

t-paul, 2019-12-25:
Indeed, that's better. Another option would be to guard with something like `assert($children > 1, "module needs at least 2 children");`

nophead, 2019-12-25:
difference() doesn't need any children, so a replacement for it should not
assert.

MichaelAtOz, 2019-12-25:
and for debugging, sometimes it is convenient to comment out various children, sometimes leaving one.

nophead, 2019-12-26:
Indeed and also sometimes some ifs that may leave only one child, for example a conditional hole.