openscad/openscad #3168: Some thing evil in rotate_extrude?

MichaelAtOz, 2019-12-23:
From #3117, various code producing self-intersections and duplicated triangles.
With 2019.05 & 2019.11.20.
```
difference (){
  translate ([5,-2,0])
    cube ([20,4,4]);
  translate ([0,0,2])
    rotate_extrude(angle=360,$fn=60)
      translate ([11,0,0])
        circle (d=3.4);
}
```

Exported, Netfabb shows; (SI's at top of circle)
[image]

Close-up;
[image]
The white triangles according to Netfabb are 'duplicate triangles', the yellow marks are where an  vertex is in the middle of an edge, hence degenerate triangles. [note the Netfabb default colour setting is not white but orange, I changed it to make sure I knew what type it was]

STL [#3117-2.zip](https://github.com/openscad/openscad/files/3996537/3117-2.zip)

Unlike @bakerjvp I have yet to get a crash.

nophead, 2019-12-23:
I suspect it is because one of the segment junctions of the torus meets the top of the cube at a slight angle and the  difference creates the normal vertices too close for the grid problem.

A tricky thing to avoid but I think if the segments junctions are wholely inside or outside the cube it would work.

MichaelAtOz, 2019-12-24:
Yes it is at the tapered segment of the torus;
[image]

So yet another grid/small-float issue. 
This is bad, that it a perfectly normal bit of code, which should not need workarounds.

MichaelAtOz, 2019-12-24:
Indeed, `rotate(2)` the `rotate_extrude()` moves the segment edge and produces a good STL.
[image]

These issues give OpenSCAD a bad reputation...

MichaelAtOz, 2019-12-24:
Closing, track via #2686 or #2991.

closed 2019-12-24 by MichaelAtOz

thehans, 2019-12-24:
> These issues give OpenSCAD a bad reputation...

Is this STL showing errors in any program besides Netfabb?  Maybe the issue is with netfabb doing some processing upon import and messing up things which were fine.

I checked that re-importing to OpenSCAD and doing CGAL ops on the STL works fine, no non manifold warnings or anything.  

Also using Meshlab and choosing "Filters -> Selection -> Select Self Intersecting Faces" doesn't pick up anything...

Unless there's any second program to corroborate Netfabb's findings, I'm inclined to blame Netfabb.

MichaelAtOz, 2019-12-24:
Hmmm.
Here is one I prepared earlier
[image]
```
// to sharpen the pillar
!difference() {
  import("pillar.stl");  
  import("sharpner.stl");
}
```
STLs produced by OpenSCAD appear clean, [pillar.zip](https://github.com/openscad/openscad/files/3996909/pillar.zip)

Via
```
//to create a pillar: pillar 1
linear_extrude (height=80)
  union() {
    translate ([0,-1.1,0])
      color ("blue")
        circle ($fn=40,d=4);
    translate ([0,1.1,0])
      color ("blue")
        circle ($fn=40,d=4);
    translate ([-2,-1.1,0])
      resize ([4,2.2,0])
        square (4);
  }
```

```
//to create sharpner 1
difference () {
  translate([-7,-7,60]) 
    cube ([13,13,20]);
    $fn=60;
  translate ([0,0,60])
    cylinder (r1=3.5,r2=0.1, h=25);
}
```