openscad/openscad #3177: Crash when hulling two circles

nyponen, 2019-12-31:
OpenSCAD is crashing when I hull two circles with $fn=4. I've attached a simplified test case. In this test case, the circle() is done within nested rotate()s that cancel each other out. If you remove the rotate()s, the problem does not occur. If you change the rotate angle to an integer, the problem does not occur. If you change $fn to any number other than 4, the crash does not occur.

[bomb19.txt](https://github.com/openscad/openscad/files/4013363/bomb19.txt)

Possibly related to [https://github.com/openscad/openscad/issues/2220](url) ?

thehans, 2019-12-31:
@nyponen Your link was broken so I edited it.  You can use the shorthand eg `#2220` in github comments to automatically link to issues and pull requests within the same project.

MichaelAtOz, 2020-01-01:
Reproduced on W7, Preview gets Exception Code:	40000015.
Google says;

> 0x40000015 means STATUS_FATAL_APP_EXIT. Your app causes an unhandled runtime exception during shutdown. Some runtime exceptions are actually handled if you don't handle them yourself, and some of these default handlers call abort(). 

2019.12.22.nightly on Debian gets;

>A critical error was caught. The application may have become unstable:
CGAL ERROR: postcondition violation!
Expr: is_ccw_strongly_convex_2( res.output_so_far_begin(), res.output_so_far_end(), ch_traits)
File: /usr/include/CGAL/Convex_hull_2/ch_akl_toussaint_impl.h
Line: 337

GUI is then hung.

MichaelAtOz, 2020-01-01:
Not hung, actually, but the progress meter was stuck, in closing it got the 'Do you want to save your changes?' dialogue.

kintel, 2020-01-04:
Closing as duplicate of #2220

closed 2020-01-04 by kintel

nyponen, 2020-01-04:
This is not a duplicate of #2220. #2220 has two different crashes in it. The first resulted from using very small dimensions. That is not what caused #3177. The second crash in #2220 no longer occurs in the current release of OpenSCAD. #3177 should be reopened.

kintel, 2020-01-04:
The second example of #2200 was indeed fixed in OpenSCAD-2019.05. Thanks for pointing that up; issue updated.
The first examplein that issue is the same bug as this; it's a problem with stability of an underlying library call. I've added this testcase to #2220, so we remember to test it once that gets fixed.