openscad/openscad #3185: OpenSCAD on Mac OS now gives occasional security warning about deprecated use of HTTP instead of HTTPS

mjparme, 2020-01-09:
This warning appears on occasion now (maybe due to Catalina upgrade?).

"You must change the feed URL (http://files.openscad.org/appcast-snapshots.xml) to use HTTPS or disable App Transport Security.

For more information:
https://sparkle-project.org/documentation/app-transport-security/"

[image]

t-paul, 2020-01-09:
Running which version of OpenSCAD?

mjparme, 2020-01-09:
> Running which version of OpenSCAD?

Sorry, I guess that could be important info, here is my veversion:

```
OpenSCAD version 2019.05 
Copyright (C) 2009-2019 The OpenSCAD Developers
```

t-paul, 2020-01-09:
Thanks, not sure why that happens. All links I can find are changed to HTTPS before that release. @kintel any idea where that's coming from (and why the release version polls the `*-snapshot.xml` file?

kintel, 2020-01-10:
That's weird. All links were changed to https ~a year ago (https://github.com/openscad/openscad/pull/2689), months before that release.

Following the link (https://sparkle-project.org/documentation/app-transport-security/), is says: "App Transport Security has other specific requirements too, so please test updating your app on 10.11 even if you already are serving over HTTPS!"

I could install Catalina in a VM for testing, but I don't really want to start on that unless I know it will work, as it's a bit of a time sink.

mjparme, 2020-01-14:
I installed 2019.12.20 (dev) and haven't seen the warning again.

kintel, 2020-01-14:
Thanks for the update. Closing as resolved

closed 2020-01-14 by kintel