openscad/openscad #3188: Feature Request: PDF Export

sbridger, 2020-01-13:
There is not (for me at least) a workable way to get a correct size paper slice from Openscad without pain. 
.
DXF export -> Online-PDF_Convertor results in vanishingly fine lines that do not print on a laser.
SVG is not directly printable, nor is PNG or anything else.

There are many cases where a 2D part design is needed and this would be useful.
It would also help reduce wastage of 3d printer plastic by allowing much cheaper easier size checking.

Export to PDF should have options (in decreasing order of importance to me) to:
 

1. Print the DXF export outline with specified, visible line width OR
 2. Print the PNG/SVG export view
 3. have an X,Y scale ruler along the edge for calibration, and/or a calibration grid
 4. print all the "echo" output on the page
 5. filename, datetime in footer
 6. add a title etc
 7. able to obey color() : https://github.com/openscad/openscad/pull/2035
 8. export() subtrees programatically and iteratively so a pdf / pdfs could contain the multiple cutting/drilling drawings of an object.  https://github.com/openscad/openscad/issues/98#issuecomment-7512305

t-paul, 2020-01-13:
Sorry, this is way to broad a feature request including features that are out of scope for OpenSCAD language (`export()`).
I'd suggest to provide a clearer example/description of the core feature for discussion.

sbridger, 2020-01-13:
Not sure what is unclear: Print the DXF outline to PDF so it can be printed on paper.

Core feature: is 1 or 2 (either works for a drill template. 1 is probably more important for laser cutter)
Others are in decreasing order of importance and optional options.

3 is fairly important as it's quite easy to print pages with "scale to fit paper" and get a slightly wrong sized print

thehans, 2020-01-13:
Line thickness is not desirable for laser cutters.  A toolpath is ideally a zero-width line, and most laser cutter software can already load DXF directly.  Adding width to DXF lines would mean that laser would do 2 passes on everything, one for each side of the line.

sbridger, 2020-01-13:
Perhaps DXF export needs to have configurable linewidth, then the online PDF convertors might be useful to make printable docs.

I would certainly prefer configurable linewidth in PDF's, as sometimes you need fine lines, and sometimes you want them easy to see under swarf and so on.

t-paul, 2020-01-14:
Talking that much about DXF is not helping much as this is just another export which will have not much in common with a new PDF export.

- the internal 2d model has no line sizes, that's probably not going to change soon
- for configurable line sizes (and other options) we need #649 or annotations that can be extracted at export time
- I'm not sure we want to bring in actual printing, if it's nicely encapsulated (with cross-platform support) in Qt it might be possible to add this, but this ticket should focus on the export first

So to limiting things a bit for an initial implementation could look like:
- Export 2D model to PDF single page portrait, no scaling, origin bottom left, Y vertical
- Have an X,Y scale ruler along the edge for calibration, and/or a calibration grid
- Header/Footer: Title, Filename, Date, Time

Needs also investigation of libraries, so far it's not looking ideal.
- libharu - MXE: yes, Debian: no, Homebrew: no (looking for new maintainer)
- libpodofo - MXE: no, Debian: yes, Homebrew: yes (not much activity, but looks maintained)
- poco-pdf - MXE: yes, Debian: yes, Homebrew: yes (nice availability, big framework, needs closer look)
- cairo - MXE: yes, Debian: yes, Homebrew: yes (nice availability, big framework)
- skia - MXE: no, Debian: no, Homebrew: no (maintained by Google)

t-paul, 2020-10-05:
Closed via #3266 / #3443.

closed 2020-10-05 by t-paul