openscad/openscad #3190: Feature request: File export

weheh, 2020-01-16:
I love OpenSCAD. It's very useful. I have been doing procedural 2D and 3D design since 1985. I have tried for years to get off OpenSCAD because of some of its limitations but I'm stuck because it works so much better for me than anything else. Bottom line, I love OpenSCAD.

I think some basic file management statements in the language could streamline the flow dramatically. It would be nice to be able to render the whole design and output the render to an STL or DXF format. Testing for existence of a file would be really useful, too. That way, the flow is for my OpenSCAD program to test whether an STL file exists. If not, then render a really complex part and write it to an STL file. Next pass ... import the STL file instead of rendering from scratch. If timestamp info were available, then I could also do currency checking on the OpenSCAD file and the STL file. 

Thanks for the great work! I love OpenSCAD.

t-paul, 2020-01-16:
Can you please explain the reason why you'd like to have file I/O? It sounds like the problem is that caching is not persistent (see #1645). It's unlikely that the language will get direct I/O control, there might be additional features via GUI and/or command line options.

weheh, 2020-01-16:
I have a complex design with features whose parameterized locations must be communicated to a downstream CNC tool for machine avoidance. I can echo coordinates and then cut and paste them into a file for use by my downstream program that prepares the GCODEs, but that's a pain. So I duplicate some of my computations in my downstream Python code that generates the GCODEs.

That's one example.

The business of persistent cache, as I describe it, is another.

Some of my designs take a long time to compile. I then have to periodically inspect the run so that when the rendering is done, I can tell the system to export an STL. This is a cumbersome delay that requires a bunch of mousing and typing. I'm lazy and clumsy and there's too much room for error and retyping here. 

As I said, I've been doing module generation, a.k.a. procedural design, for decades. A procedural design system always fits into a flow. Period. If it can't communicate to the flow except through a human, the tool's effectiveness is hobbled.

weheh, 2020-01-16:
Permit me to clarify further. I use OpenSCAD to make several unique views and versions of the same design. One run of the generator could result in 25 different DXF files and 10 different STL files. These are all intended for other tooling downstream. 

The file names follow a descriptive naming convention produced automatically using the str() command. File names can get intricate. Doing this all by hand after a tweak in the design turns out to be a big pain and saps some of the power behind the procedural approach.

MichaelAtOz, 2020-01-16:
@weheh are you using [command line](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Using_OpenSCAD_in_a_command_line_environment)? You can output echo with `-o name.echo`, unfortunately without persistent cache it may need to re-render because it can't output both STL & echo in the same run. But if you structure you code to do the echo'ed calculations with out generating geometry, then have a script format the echo output into assignments in a .scad file, you can then `include<>` those values into another run to make the geometry.

The problem with your proposal is OpenSCAD is a markup language not a procedural one, there is no 'flow'

weheh, 2020-01-16:
Thanks for the -o name.echo suggestion. I haven't looked at the command line in practically forever.

The fact that there's order dependence in OpenSCAD's various transformations makes it procedural to me -- first do this, then this, then that. The fact that OpenSCAD is part of my design methodology makes it part of a flow. Getting people to integrate OpenSCAD into various flows makes them dependent on it and therefore increases its chances of being relevant for a very long time. If it's standalone, then it's more easily replaceable by best-in-class tools.

weheh, 2020-01-16:
Aside: I guess the STL part doesn't matter that much at the moment since OpenSCAD can't read back in and display the STL that it wrote, in my case. No error messages rendering, writing the STL or reading it. However, STL file is good and can be imported to other tools just fine. I may need to file another issue report.

MichaelAtOz, 2020-01-16:
> first do this, then this, then that

No, it just IS, each of those statements is just a description of the geometry, it doesn't iterate.
Each OpenSCAD operation in your overall 'flow' is just a description of a static picture at that point.

The only 'flow' like thing could be animation, but each slice of the animation is a static picture.

BTW is anyone else having GitHub not working for Preview, or the various icon buttons AA B i etc?

MichaelAtOz, 2020-01-16:
> Aside: I guess the STL part doesn't matter that much at the moment since OpenSCAD can't read back in and display the STL that it wrote

Do you mean import("your.stl") is not working, producing errors or bad geometry?
Raise another issue, post your STL, I'll have a look at it if you want.

weheh, 2020-01-16:
You're making me think of the famous quote, "It depends upon what the meaning of the word 'is' is. " Anyway, in my semantics, non-commutative languages are procedural.

weheh, 2020-01-16:
Sorry, I can't post the STL. It's confidential. This is an intermittent problem and seems to affect my most complex designs, so I can't easily put together a test case. I have abutting geometries so I think it's related to that. For the time being, I'm able to limp along. Generally, what happens, is the STL file reads in but then either displays partially or not at all.

nophead, 2020-01-16:
Well that is simply incorrect. Functional languages are non procedural but
that does not mean the order of applying operators does not change the
result.

In general OpenSCAD cannot import STL files s it exports because if there
are any vertices closer than the resolution of a float it will fail.

MichaelAtOz, 2020-01-17:
> You can output echo with `-o name.echo`, unfortunately without persistent cache it may need to re-render because it can't output both STL & echo in the same run.

@t-paul this seems to be an unfortunate limitation, would it be worth pursuing an option to allow the echo to be captured concurrent to geometry export?

weheh, 2020-01-17:
Whatever, I'm not here to debate semantics. Seems like this is a deep limitation and related to the other problem I alluded to when I abut features that also follow a curved surface and then it won't render. I read another thread that ran into the same problem. Seems like a deep flaw. I think I've churned many hours trying to deal with this floating-point issue.

kintel, 2020-01-17:
@MichaelAtOz @weheh Having multiple `-o` options could be done. If this is an important feature, please open a separate github issue.

kintel, 2020-01-17:
Being able to read back STL files is a blocker of implementing persistent caching using STL files (or any format based on floats). Just not sure how we should tackle that.

laird, 2020-01-17:
Couldn’t you achieve the ‘STL caching’ in effect by wrapping the OpenSCAD that generates a part in a script, with logic like ‘if the STL file doesn’t exist or is old, invoke OpenSCAD to render the STL. Then pass the STL downstream to the rest of your process?’ I do this sort of thing using bash and Python scripts quite heavily to generate parametric designs in a multi-step production process, and perhaps that approach might work for you?

kintel, 2020-01-17:
@laird Yeah, using Makefiles is a common way of achieving that. We even have a way of generating dependencies for Make (https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Using_OpenSCAD_in_a_command_line_environment#Makefile_example), but it's not too well maintained.

t-paul, 2020-01-17:
I agree that multiple `-o` exports from command line could be useful and are probably not too difficult to add. IIRC that even came up not long ago (on the forum?).

For the other topics it seems the discussion is not going anywhere. There's not much point in just calling everything a bad limitation or even "hostile".
The reason for trying to understand the workflow is that we can't support just every kind of workflow and special case. We need to find some common ground which is possible to support with a very small team.
So please lets step back a bit and see if we can define some features/extensions that cover at least some of the areas mentioned while still being commonly useful.

weheh, 2020-01-17:
@laird & @kintel: yes, you are correct, a makefile or an upstream python script could do the currency checking. Probably doesn't belong in OpenSCAD given what I understand of its architecture. Anyway, having a command line option to export both STL and text would open room for more automation in my flow, so seems like a worthy enhancement. Since I haven't perused the enhancement list, I don't know how close to the top it would float. I leave that to others to decide. Thanks for the discussion. And again, thanks for OpenSCAD.

thehans, 2020-01-18:
> I agree that multiple `-o` exports from command line could be useful and are probably not too difficult to add. IIRC that even came up not long ago (on the forum?).

That might have been me blabbing about it on IRC.
I actually got it working a few days ago, its a fairly code change needed... I just hadn't gotten around to committing changes until now.  See PR #3191

t-paul, 2020-01-20:
@thehans quite possible.

So to wrap up, the suggestions based on the discussion are
1) Get PR #3191 in which would allow to export multiple formats at once from command line
2) Point to the Makefile support which may need some love, but changes would need a separate issue describing what is needed to improve processing via `make`

If there's no additional topics for now, I'd close this issue.

PRouzeau, 2020-01-30:
Workflow with part generation and producing of echo files was discussed in #1999 issue

closed 2020-04-26 by t-paul