openscad/openscad #3194: Syntax error in tutorial

K2AAE, 2020-01-20:
translate([5,0,10 - 0.001]) throws syntax error

cube([60,20,10],center=true);
Z = 10 - 0.001;
translate([5,0,Z])
    cube([30,20,10],center=true);
cylinder(h=3,r=8);
this works but that leaves the tutorial broken...

kintel, 2020-01-20:
Looks like the tutorial uses the `–` character (unicode 0x2013) instead of `-` (ascii 0x2d).
This would likely need to be changed for all examples in the tutorial.

rcolyer, 2020-01-20:
This should all be fixed now.  10 minutes of clicking with a perl script to swap them all resolved it.

MichaelAtOz, 2020-01-20:
Where is this tutorial?

rcolyer, 2020-01-20:
https://en.wikibooks.org/wiki/OpenSCAD_Tutorial

t-paul, 2020-01-27:
I checked most of the code and it seems @rcolyer did already catch all the places where that issue happened. I just fixed another typo. I think with that we can close the issue.

closed 2020-01-27 by t-paul