openscad/openscad #3199: Preview Bug with cylinder() difference() and resize()

CryptKid, 2020-01-27:
In an object i'm designing, i have a roof consisting of the of two cylinders and a rotated cube.
When the edge cutten of by the cube is in sight, the entire roof develops holes or vanishes. From other perspectives it looks just fine. Sourcecode is:
```openscad
difference() {
    translate([0,42-28,2]) resize([0,0,11]) rotate([-90,0,0]) difference() {
        cylinder(140.5,40,40);
        translate([0,0,-0.1]) cylinder(146.2,35,35);
        translate([-40,0,-0.1]) cube([80,40.1,190.2]);
    }
    translate([-40,0,0]) rotate([-45,0,0]) cube([80,10,30]);
}
```
Also after being rendered, the roof looks fine from all perspectives, that's why i suspect a bug

The attached pictures are all made without any change to the source, only by rotating
[image]
[image]
[image]

After refactoring it to
```openscad
translate([0,42-28,2]) resize([0,0,11])  rotate([-90,0,0])difference() {
     
        cylinder(140.5,40,40);
        translate([0,0,-0.1]) cylinder(146.2,35,35);
        translate([-40,0,-0.1]) cube([80,40.1,190.2]);
    
    translate([-40,0,-25]) rotate([85,0]) cube([80,40,60]);
}
```
Everything works fine but it makes it a lot harder to find the right rotation for the cube

nophead, 2020-01-27:
Yes it looks like resize() needs a convexity parameter to preview properly, like render().

t-paul, 2020-01-27:
Yeah, not really possible to say with the tiny screenshots - see also [FAQ entry](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/FAQ#Why_are_some_parts_(e.g._holes)_of_the_model_not_rendered_correctly?).

nophead, 2020-01-27:
The FAQ says set convexity but the bug is that resize has no convexity parameter. I think it needs one as it behaves like render().

t-paul, 2020-01-27:
Right, found it in https://github.com/openscad/openscad/blob/master/src/cgaladv.cc#L70 which has convexity but does not evaluate it for `resize`.

commit 02ca2188 (2020-02-16): Merge pull request #3224 from openscad/resize-convexity

Add convexity to resize(). (fixes #3199).
commit 32df2aaf (2020-02-16): Add convexity to resize(). (fixes #3199).
closed 2020-02-16 by t-paul