openscad/openscad #3201: Feature Request: extract usable data from SVG

RichyT, 2020-01-29:
I'm keeping this kind-of nonspecific as I'd like to see some discussion. What I'm thinking is that when we import an SVG, it would be nice to read back some information from the SVG file we're importing and be able to use that in the OpenSCAD script. For example, an item in the SVG might have a tag 'openscad.thickness="5"' where we could then do

linear_extrude(height = thickness) import(file1);

Now, I can see a couple of the issues here but as I say, I'm keeping it loose to see what discussion comes up.

t-paul, 2020-01-29:
I think that is out of scope for OpenSCAD. I'll leave the ticket open for a bit, but consider this is not a discussion board. Open questions are better placed in the forum.

t-paul, 2020-05-11:
I'm closing this for now as it's a bit too unspecific. It might come up again once we solved the data structure topic which would allow importing additional data.

closed 2020-05-11 by t-paul