openscad/openscad #3209: ERROR: CGAL error in CGALUtils::applyBinaryOperator intersection: CGAL ERROR: assertion violation! Expr: e_below != SHalfedge_handle() File: /usr/include/CGAL/Nef_3/SNC_FM_decorator.h Line: 427

traverseda, 2020-02-05:
Failure to render mesh with multiple linear-extruded intersections. Preview is fine.

```
 width=60;
 sides=7;

!intersection(){
    linear_extrude(height = 100,  convexity = 10, $fn=20,twist=180)circle(d=width,$fn=sides);
    linear_extrude(height = 100,  convexity = 10, $fn=20,twist=360){
        translate([-2.5,0,0])square([5,width]);
    }
}
```

OpenSCAD version 2019.05

MichaelAtOz, 2020-02-05:
Reproduced on W7 2019.05 & Debian 2020.01.04.nightly.

nophead, 2020-02-05:
I think it is because such a tight twist causes self intersections unless slices is set to a suitable high value in the second linear_extrude, e.g. 50.

traverseda, 2020-02-06:
Awesome, dice tower is a go.

```
 width=60;
 sides=7;
 
linear_extrude(height = 100,  convexity = 10, $fn=20,twist=180)
difference(){
    circle(d=width,$fn=sides);
    circle(d=width-5,$fn=sides);
}

intersection(){
    linear_extrude(height = 100,  convexity = 10, $fn=20,twist=180)circle(d=width,$fn=sides);
    linear_extrude(height = 100,  convexity = 10, $fn=40,twist=360*2)translate([-2.5,0,0])square([5,width]);
}

cylinder(d=5,h=100);
```

Not sure how one would even fix this, aside from providing a more descriptive error message.

kintel, 2020-02-09:
This is most likely a duplicate issue. If someone hunts down the duplicate, it could make sense to copy this example over and close this one.

MichaelAtOz, 2020-02-09:
I had a look, this seems to be the only CGALUtils::applyBinaryOperator **intersection**, plenty with **union**. 

However, having looked at many issues and my own history, I think `linear_extrude` is likely at fault.

nophead, 2020-02-09:
Yes linear extrude with a big twist of something thin always causes self intersections unless slices is set to a suitable value. The reason is the tessellation of twisted quads causes creases along the diagonals that can be convex enough to reduce the thickness to less than zero.  I think if it picked the diagonal that made it concave then both sides would move away from each other but the result is ugly anyway unless you set slices to make the quads roughly square.