openscad/openscad #3214: Next Release

kintel, 2020-02-10:
We want to publish a new release soon. This issue functions as a release checklist.

* [ ] Cut a RC branch and build RC binaries
* [ ] Update copyright year to 2020 in all place visible within the app
* [ ] Update src/AboutDialog.html with changes to libraries, people etc.
* [ ] Update doc/openscad.1 with new cmd-line arguments
* [ ] Update changelog
* [ ] Prepare news entry

thehans, 2020-02-10:
Oh, speaking of changes to AboutDialog people, I'd like to display my full first & last name while we're at it. :smile: 
"Hans Loeblich"

Just a small thing I didn't bother bringing up before, since the current listing "Hans L" was how I had originally written on my github account.  I've since changed that too.

sk8nfool, 2020-04-13:
It has been a couple of months. Is this release still coming?

t-paul, 2020-04-13:
Yes

mholiv, 2021-01-15:
I'm not sure if this is the right place but I found two regressing in RC5 that I documented here:
https://github.com/openscad/openscad/issues/3614
https://github.com/openscad/openscad/issues/3615

I wanted to let people know before the RC process was finished.

t-paul, 2021-01-15:
RC is pretty much finished unfortunately. I'm not even sure there's much to do about those 2 cases as those are likely platform/distribution related. We'll see about the details in the specific issues...

t-paul, 2021-02-07:
Closed via https://github.com/openscad/openscad/releases/tag/openscad-2021.01

closed 2021-02-07 by t-paul