openscad/openscad #3217: Allow resize of console window

bburdette, 2020-02-12:
I'm unable to expand the console window, which makes viewing the program output inconvenient - I can only see a couple of lines of text at a time.  I can grab the console window and drag it around, but it goes right back where it was, at the same size that is was before.  There does't seem to be any kind of handle around to grab for resizing the panel.

This is OpenScad v 2019.05.

nophead, 2020-02-12:
I just hover over the line that is above the word "Console" and the cursor changes to two parallel lines with arrows and then left click and drag moves the division between the two windows. This is the same version on Win7.

bburdette, 2020-02-12:
wow, how'd I miss that?  I don't have a line over 'console', but the cursor does change there.

nophead, 2020-02-12:
It looks like this on my PC
[image]

bburdette, 2020-02-12:
I'm on linux, i get this: 
  
[image]

t-paul, 2020-02-12:
This depends on the theme defined by the system, not an OpenSCAD issue.

E.g. flatpak version on my Debian looks like:
[image]

closed 2020-02-12 by t-paul