openscad/openscad #3218: Customizer not usable with floats when value is .0

LordRofl, 2020-02-14:
Hello,

i have a model where i need to change the value smaller then 1 and that is right now impossible with the customizer when the value in the script ends with .0

My workaround was to change all variables from:

testvar = 0;
to
testvar = 0.0;

But that does not work, because openscad changes the float to a integer on its own.
A bit frustrating because to change settings in the script to something else then .0 is not the solution for me. 

Hope that get fixed :)

LordRofl, 2020-02-14:
To be clear i have a variable in the script set to 0 and right now its not possible to change the variable with the customizer to 0.5.

I hope someone can give me a solution for this

t-paul, 2020-02-14:
OpenSCAD does not have different types of numbers, so you have to give it some additional hint by providing a range, e.g.  `// [0:0.5:100]` should help, although that might change the GUI representation but that's a different topic...

See also https://customizer.makerbot.com/docs

LordRofl, 2020-02-14:
Ohh thats great. I'm new to the software and for me that was a odd behaviour.

Is it possible to keep the comment that the line also have? // [0:0.5:100] HERE is more text

t-paul, 2020-02-14:
No, as the docs describe pretty much at the beginning it has to be
```
// variable description
variable name = default value; // possible values
```

LordRofl, 2020-02-14:
Yea i read it all and iam amazed! I already change the script with Parameter Tabs and all. Thank you very much. I can close this now in peace :D

closed 2020-02-14 by LordRofl