openscad/openscad #3219: Feature Request: Exploded Color SVG's

Metzen2k, 2020-02-14:
When using OpenSCAD for laser cutting (which a lot of people do) an SVG is exported. The SVG is all one big path, so in Inkscape, etc you can't just select a hole or a letter and give it a different stroke color to tell it to cut, score, or engrave, or for different power or speed settings.

I created a simple PHP script that takes an OpenSCAD SVG and explodes the path into it's individual parts and removes the fill. It is then very easy to edit in just a few seconds to apply different stroke colors (settings for the laser cutter) to different elements in the SVG.

[breakApartOpenScadSvg.php.zip](https://github.com/openscad/openscad/files/4207253/breakApartOpenScadSvg.php.zip)

This could be done within OpenSCAD and offered as a separate export, "Exploded SVG" or whatever, in addition to the current SVG (so we don't break the existing workflow for anyone).

This would pave the way for the next part: color SVG's. So when you create a circle, for instance, you can pass a parameter for color that ends up in the exploded SVG:
circle(d=5, svgColor=red);
circle(d=10, svgColor='#FF0000');

That would be a fairly straight forward way to make OpenSCAD much more powerful and much easier for people using OpenSCAD for laser cutting, without changing things for anyone not using the feature.

Thanks!

Metzen2k, 2020-02-14:
To give more flexibility the parameters could be separate for fill and stroke:
circle(d=5, svgStrokeColor=red, svgFillColor=black);

t-paul, 2020-02-14:
Adding special tags specific to export formats is not going to scale very well and would require handling all combinations on every CSG operation.

It would be helpful to collect the details on what would be required for better laser cutter support in one ticket, so I'll close this one as duplicate of #1364.

closed 2020-02-14 by t-paul