openscad/openscad #3227: Something annoying about "C-like" for

RonaldoCMP, 2020-02-18:
This issue has been discussed previously in a forum thread:
http://forum.openscad.org/Something-annoying-about-C-like-for-tp28292.html

The C-like for has the following general form:

[for( initialize(vars);
    test(vars);
      iterate(vars) ) output(vars) ]

The outputs always happen after each test and the iteration ends just after a test before any output. So, a better understanding of the control structure is realized by the schema:

[ for(  initialize(vars),
         i = 0;
     i < 8;
       (output(i,vars))
       i = i+1,
       iterate(i,vars) ] 

where I consider a simple iteration with a counter limit test. In that example, we will have 8 outputs and 8 iterations besides the initialization. The initialization may produce an output but the last iteration never does: it is a useless iteration. When the iterations include very demanding processes we need to do tests in the iteration steps in order to avoid them in the last iteration. The same happens when the last iteration may fire warnings or errors due to bound overpass. Besides, when the test includes checking variables other than the iteration counter we need to create additional variables to allow a output test to recognize that the next iteration will be the last one.

All that could be avoided if the output was done before the test. I see two possibilities here: the output is done always before the test or it is done just after the iteration part. In this last case (that seems natural because it is the syntactic position of the output code), there will be no output after the initialization. These alternatives have their own merits and demerits but any of them seems to lead to simpler codes than the current scheme.

t-paul, 2020-02-18:
As shown in the manual, this is just short way of writing
```
function f(a, b, ...) =
    condition
    ? concat([expr], f(nexta, nextb, ...))
    : [];
  f(inita, initb, ...)
```
Simply changing the behavior is not possible anymore, however there's the option of having an additional loop that handles those cases. What are other languages doing in that case?

RonaldoCMP, 2020-02-18:
Very interesting, Torsten. I was not aware of that equivalence. It helps me
to realize that the two alternative behaviors I had proposed can be
achieved with an equivalent recursion:

function f(a, b, ...) =
    condition
    ? concat([expr], f(nexta, nextb, ...))
    : [expr];

f(inita, initb,...);

Based on that equivalence I guess that better codes with "C-like for" may
be written for complex cases.

Thank you.

t-paul, 2020-02-20:
So I guess we would need to find a nice name for that :-). That said, it would be even cooler if there would be a way to define that kind of functions in user space, similar to modules using `children()`.

RonaldoCMP, 2020-02-20:
I can't say I have understood your cooler alternative. But it would be nice
to have an iteration syntax, named 'for' or something else, able to cover
those cases that seem to be more frequent. I could not devise any way to
get it without either filtering the last iteration or repeating it at the
output.

RonaldoCMP, 2020-03-07:
> So I guess we would need to find a nice name for that :-). 

I have two suggestions for new iterations that come up to my mind when I recently ported to OpenSCAD a 2D convex hull code written in C. The first one is an expression:

`until( <named_expressions>; <cond>; <expression> )`

that returns the evaluation of `<expression>` at the end of the last iteration; the iterations evaluate the list of expressions `<named_expressions>` mediated by the test `<cond>`; when it fails, `<expression>` is evaluated and returned. A simple example:

`function sum(p) = let( s=0, i=0) until(  s= s+p[i], i=i+1; i>=len(p); s );`

I guess `until` is equivalent to something like:

`function until(a, b, ...) = cond ? <expression> : until(newa, newb, ...);`

The other iteration is a list comprehension expression with a syntax like:

`[ while( <cond> ; <named_expressions>) <expression>]`

In this case, while the `<cond>` is true, `<named_expressions>` and `<expression>` are evaluated; each evaluation of `<expression>` is inserted in the list. A simple example:

`function accumulated_len(p) = `
                    `[let( i=0, al=0 ) while( i<len(p) ; al = al+len(p[i], i = i+1 ) al ];`
`p = [ [1], [1,2,3], [1,2] ];`
`echo( accumulated_len(p) ); // ECHO: [1, 4, 6]`

The equivalence of 'while' would be:

`function f(a, b, ...) =`
`condition`
` ? concat([expression], f(nexta, nextb, ...))`
 `: [expression];`

thehans, 2020-03-09:
I know I've encountered these sort of difficulties with c-style for in the past, but I can't find a good example in my existing scripts at the moment.  

Anyways it seems like the main point is just accessing the values from that last iteration.

My suggested solution would be adding an optional `finally` clause in the output section of a c-style list comprehension, which is only evaluated and output on that last iteration.
`[ for (<init>;<condition>;<next>) <expression> finally <last_expression> ]`

Also since I think in many cases only the last value is desired, it should be allowable to have finally, without the "main" expression:
`[ for (<init>;<condition>;<next>) finally <last_expression> ]`

And of course the existing style without any "finally" would behave the same as before:
`[ for (<init>;<condition>;<next>) <expression> ]`

nophead, 2020-03-09:
Yes I think that would be a neat fix to my one and only use case so far. It would look a bit odd though because expression and last_expression would be the same variable. I would just stop the loop one step earlier.
```
        rotations = [for(i = 0, rot = fs_frame(tangents);
                         i < len;
                         i = i + 1,
                         rot = i < len ? rotate_from_to(tangents[i - 1], tangents[i]) * rot : undef) 
                             rot],
```
Would become:
```
        rotations = [for(i = 0, rot = fs_frame(tangents);
                         i < len - 1;
                         i = i + 1,
                         rot = rotate_from_to(tangents[i - 1], tangents[i]) * rot) 
                             rot finally rot],
```

thehans, 2020-03-09:
> I would just stop the loop one step earlier.

That's only viable for the most trivial type of conditions though. A loop exit condition could be comprised of compound statements or result of a function call etc that might not be so easy to predict.  The type of calculation to compute Mandlebrot set comes to mind (not that I've ever tried implementing it in OpenSCAD, just an example of a type of condition where I don't think you could easily just subtract one).

thehans, 2020-03-09:
Sorry, my comment above was a little confused and maybe not relevant, I guess I misread or misinterpreted your point at first glance chris.

RonaldoCMP, 2020-03-09:
> Hans Loeblich <notifications@github.com> wrote:
> I know I've encountered these sort of difficulties with c-style for in the past, but I can't find a good example in my existing scripts at the moment. 
> Anyways it seems like the main point is just accessing the values from that last iteration.

Both are needed as we can see by the following complete code to find the convex hull of a set of 2D points:

```
function convex_hull(p) =
// based on: 
// https://www.hackerearth.com/practice/math/geometry/line-sweep-technique/tutorial/
  let( compare = function(a, b) a.x==b.x ? a.y-b.y : a.x-b.x )
  let( is_ccw  = function(a,b,c) cross(a-c,b-c)>=0 )
  let( backtracking = function(p,h,t,m)
                        m<t || !is_ccw(h[m-2], p, h[m-1]) ?
                          m
                        : backtracking(p,h,t,m-1) )
  let( n = len(p), 
       p = qsort(p, comp=compare) ) // compare sorting
  // lower hull
  let( lh = [ for(  i = 2,
                    k = 2, 
                    h = [p[0],p[1]];
                  i <= n;
                   // safeguards
                    k = i==n ? k: backtracking(p[i],h,2,k)+1,
                    h = i==n ? []: [for(j=[0:1:k-2]) h[j], p[i]], 
                    i = i+1
                 ) if( i==n ) each h ] )
  // concat lower hull points with upper hull ones
  [ for(  i = n-2,
          k = len(lh), 
          t = k+1,
          h = lh; 
        i >= -1;
          // safeguards
          k = i<0 ? k: backtracking(p[i],h,t,k)+1,
          h = i<0 ? []
              : i==0 ?
                  [for(j=[0:1:k-2]) h[j] ] // p[0] was already in lh
                : [for(j=[0:1:k-2]) h[j], p[i]] , 
          i = i-1
       ) if( i==-1 ) each h ]
 ;
```
I need only the last iteration result in all loops. Besides, I needed to place safeguards to avoid illegal operations in backtracking() in the last useless iterations of the C-like fors. That safeguards would still needed with the "finally" predicate. And tests may not be needed but opportune when the last iteration includes expensive computations.

To write that algorithm in tail recursive form would be convoluted, inefficient and unreadable.

RonaldoCMP, 2020-03-09:
I have deleted my last comment because I realized belatedly that the 'finally' predicate dispense the safeguards of the last iterations. Remains to find a good syntax for the cases where all iterations results are needed.

nophead, 2020-03-09:
I thought ```finally``` does allow you to use the last iteration, if I understand it correctly. I.e. in my example I terminate the loop one short but the last iteration is calculated anyway and ```finally``` allows me to output it.

RonaldoCMP, 2020-03-09:
That is not really clear in Hans's proposal.

nophead, 2020-03-09:
Hmm, after reading it again I am not sure either.  I don't want a different expression output on the last iteration, I want the same expression output one more time when the test fails.

RonaldoCMP, 2020-03-09:
I guess we need something else to deal with the cases where in all iterations are useful.

nophead, 2020-03-09:
I think it depends on what you consider an iteration. The original problem is the iteration code is executed and then the test fails, so the results are computed but can't be used. If the test is true for the number of items you want to output then the iteration does one more and can go off the end of arrays, etc. It is also wasted computation.

If ```finally``` lets you use the results of the last iteration then the test can stop one step earlier and so the last iteration is a valid one and its value is used. A bit strange though.

RonaldoCMP, 2020-03-09:
Strange is how the current C-like for operates. I can't see any use case where it would be preferable to have access to the results of all iterations.

thehans, 2020-12-28:
Going by the example of the recent issue which was linked here:

```OpenSCAD
echo(sum([1,2,3])); 
function sum(x) = [for(i=0, s = x[0]; i<len(x);i=i+1, s=s+x[i]) if(i==len(x)-1) s][0]; 
```
  from @Parkinbot's https://github.com/openscad/openscad/issues/3557#issue-775434592

The issue is that `i` is being incremented before `s` is updated, and so the last iteration has `s` reading an undefined value (out of bounds) and emitting a warning (even though its not "used").

One solution under the current RC of OpenSCAD could be this admittedly ugly bit of code:
```OpenSCAD
function sum(x) = [for(i=0, s=0; i<=len(x); s = i<len(x) ? s+x[i] : s, i=i+1) if(i==len(x)) s][0];
``` 
Here, `i` is incremented *after* updating `s`, but the test condition then has to been changed to `<=` to add an extra iteration, and `s` requires a redundant bounds check within its assignment.

The way my `finally` clause suggestion would work in this case would be very straightforward:
```OpenSCAD
function sum(x) = [for(s=0, i=0; i<len(x); s=s+x[i], i=i+1) finally s][0];
```
Note that:
 - the order of assignment still must be that `i` is incremented after `s` is accumulated
 - the test condition can go back to being `<`, much more sensible
 - in this example, the main output expression is blank/omitted, since we only care about the last value
 - `if(i==len(x)-1)` is no longer needed to get just the last element
 - there's no redundant check needed on `s`, as long as `i` is being incremented after it
 - its possible to handle an empty array since `s` is initialized independently of `x`

And the recursive equivalent of this would be:
```OpenSCAD
function sum_(x, s, i) =
    i < len(x)
    ? concat([/*empty main expresssion*/], let(s = s + x[i], i = i + 1) sum_(x, s, i))
    : [ /*finally*/ s];

function sum(x) = sum_(x, 0, 0)[0];
```