openscad/openscad #3229: projection() crashes after Boolean op

Parkinbot, 2020-02-18:
While this code works on OpenSCAD 2019.12.21.ci4161 on Windows 8.1 (64bit):
```
projection()
  cube([1,1,0]);  // malformed
```
the following code crashes.
```
projection()
intersection()    // crashes also on difference()
{
  cube([1,1,0]);  // malformed
  cube();    // any object
}
```
Looks to me like projection() has a problem with a "empty" object returned from CGAL

tbharathchandra, 2020-03-08:
yeah, the error is a segmentation fault, which i think is caused by the CGALUtils::createPolySetFromNefPolyhedron3 in the process of accessing Nef polyhedron of the childnode.

tbharathchandra, 2020-03-11:
I think the condition `if (this->x > 0 && this->y > 0 && this->z > 0 && !std::isinf(this->x) && !std::isinf(this->y) && !std::isinf(this->z))`  in PrimitiveNode::createGeometry() method an empty nef_polyhedron is created. and the application crashes when CGAL_forall_halffacets method in createPolySetFromNefPolyhedron3 tries to access that empty nef_polyhedron.

commit aecbb0b5 (2020-04-05): Fix crash with projection() (fixes #3229)
closed 2020-04-05 by t-paul