openscad/openscad #3230: use inotify (or equivalent) for file modification notification

ijackson, 2020-02-18:
Currently, when the auto-reload feature is enabled, we poll every 200ms.  This needlessly uses power - particularly relevant on laptops.  Many operating systems nowadays support mechanisms for having the kernel notify you of file changes.  It would be good to use these instead, where available.

Open questions as far as I can see them:
 - maybe there is some portable wrapper library that will save us writing abstractions over inotify, kevent, etc. etc.
 - I wasn't able to easily follow the file reading and opening code to see where the list of files is or how to add them to a watch list.

nophead, 2020-02-18:
There isn't a global list of files as such. Each FileModule object, which corresponds to an OpenScad file, holds a list of files it uses and a list of files it includes.

It attempts to check the modtime of each of those files in a tree expansion, which explodes, so the mod time is cached so that each file is only checked once.  I added this here: https://github.com/openscad/openscad/blob/master/src/StatCache.cc. 

The poll doesn't use significant CPU time on my Win7 laptop, even with hundreds of files, because the OS caches the file data and doesn't actually read it from disc.

If you implement it via events please ensure it works on all OSes and file systems, in particular networked shares and GoogleDrive.

ijackson, 2020-02-18:
Chris writes ("Re: [openscad/openscad] use inotify (or equivalent) for file modification notification (#3230)"):
> There isn't a global list of files as such. Each FileModule object, which
> corresponds to an OpenScad file, holds a list of files it uses and a list of
> files it includes.
> 
> It attempts to check the modtime of each of those files in a tree expansion,
> which explodes, so the mod time is cached so that each file is only checked
> once. I added this here: https://github.com/openscad/openscad/blob/master/src/
> StatCache.cc.

Thanks for the pointers.

> The poll doesn't use significant CPU time on my Win7 laptop, even
> with hundreds of files, because the OS caches the file data and
> doesn't actually read it from disc.

Well, yes, but even waking up just to make a half dozen syscalls uses
power; it can stop a cpu going into a deep sleep mode.  People trying
to optimise laptop battery life have spent quite some effort getting
rid of periodic wakeups.

> If you implement it via events please ensure it works on all OSes and file
> systems, in particular networked shares and GoogleDrive.

I doubt it will be possible to implement this feature for all OS's but
it should be possible to fall back to polling where it's not
supported, so that nothing gets worse.

Regards,
Ian.

-- 
Ian Jackson <ijackson@chiark.greenend.org.uk>   These opinions are my own.

If I emailed you from an address @fyvzl.net or @evade.org.uk, that is
a private address which bypasses my fierce spamfilter.

l29ah, 2020-10-11:
Probably https://doc.qt.io/qt-5/qfilesystemwatcher.html is the way to go.

nophead, 2020-10-11:
The documentation says that on some systems there is a limit of 256 files. I probably have more files than that in my projects.

Does it work on networked file systems?

l29ah, 2020-10-11:
It only asks the kernel for the file change events (and polls when the kernel lacks such an interface, but it is present on Linux and Windows), and it's up to the networked file systems' implementors to communicate the updates to the OS kernel. I don't know whether yours do that. Linux NFS doesn't, for example. So we should preserve the polling logic for the use cases like yours, but i think it should be opt-in.