openscad/openscad #3231: Free rotation of the 3D view via keyboard

DLFW, 2020-02-20:
The 3D view can be zoomed and brought to some basic positions via keyboard shortcuts. But it cannot be freely rotated.

**Request:** Add keyboard shortcuts for rotating the 3D view around all three axes and for each in both directions.

Optional: Provide shortcuts for small and big rotation steps, maybe 1° and 8°.

This request is a follow-up for #2977, which has been decided to be used as tracker for “configurable” shortcuts, only. (By @t-paul in https://github.com/openscad/openscad/pull/3226#issuecomment-589074106_.)

t-paul, 2020-02-20:
The comment is on the PR and separate review of smaller things helps with limited time, but I guess it's even better to have a specific feature request too.

thehans, 2020-02-21:
Something else that might be worth thinking about is the possibility of combining the shortcut key assignment functionality with input device configuration which already exists as "Buttons" tab in preferences.  
So that some of these new actions could be be assigned to joystick buttons, and shortcut keys could be set alongside buttons through some common interface.

t-paul, 2020-02-21:
Indeed. If we go with the additional actions, it should be just a matter of registering those with the input driver manager.

jordanbrown0, 2023-11-18:
Let's generalize this to rotate, pan, and zoom; they all seem to go together and it allows closing #4448 as a duplicate.