openscad/openscad #3234: Automatic brace insertion feature for editor

tbharathchandra, 2020-02-24:
Automatic brace insertion is one of the feature that almost every editor has. Yeah, few people may not like this feature but by placing it as a choice in editor preferences will be a good option.

tbharathchandra, 2020-03-12:
Seems like Qscintilla does not provide a direct method to provide this. Having a slot for the signal textChanged() can probably do this job.

Sharma-Hrishabh, 2020-03-20:
Hey, @tbharathchandra , are you going to include the automatic brace insertion feature during your GSoC period? As this looks under my selected idea's scope, So I was thinking to add that. It would not be a problem if you want to do it, in case you, have already made some progress :).

tbharathchandra, 2020-03-20:
Hey, @Sharma-Hrishabh , I am not working on it currently. You can go ahead.

Sharma-Hrishabh, 2020-03-20:
Thanks @tbharathchandra :)