openscad/openscad #3237: Unexpected behavior with for loops with floating point limits.

NateTG, 2020-02-26:
version 2018.07.21 

The for loop behavior is strange when switching between the integer and float values for l_steps here.  It seems like there's some kind of accumulation happening with nested for loops in the [] syntax.

```
l_steps=173;
//l_steps=173.205;
a_steps=19;

t_hull=[for(a=[0:a_steps]) 
          [for(step=[0:l_steps])
             [a,step,0]
          ]
       ];

polyhedron(
   points=[for(a=[0:a_steps]) for(b=[0:l_steps]) t_hull[a][b]+[a/5,0,0]],
   faces=[for(a=[0:a_steps-1]) for(b=[0:l_steps-1])
      [b+a*(l_steps+1),b+a*(l_steps+1)+1,b+(a+1)*(l_steps+1)]
   ]
);
```

nophead, 2020-02-26:
Never use floating point indexes in loops. Always use an integer and scale it. Floating point doesn't work on computers because it is done in binary and so most decimal fractions are infinitely recurring numbers and so give errors when accumulated because they can't be represented exactly in binary.

NateTG, 2020-02-26:
> Never use floating point indexes in loops.  ...

Does that have anything to do with whether this is appropriate or expected behavior from OpenSCAD?

nophead, 2020-02-26:
In any programming language on a digital computer that uses binary floating point you won't get accurate results accumulating 173.205. It isn't an exact number in binary.

NateTG, 2020-02-26:
> In any programming language on a digital computer that uses binary floating point you won't get accurate results accumulating 173.205. It isn't an exact number in binary.

What does "accumulating 173.205" mean?  The loops should all be starting at 0 and incrementing by 1.

nophead, 2020-02-26:
Yes the loop will stop at 173 and t_hull will be exactly the same but you multiply by l_steps when calculating the vertex numbers in the polyhedron, so how do you expect those to be integers?

NateTG, 2020-02-26:
> Yes the loop will stop at 173 and t_hull will be exactly the same but you multiply by l_steps when calculating the vertex numbers in the polyhedron, so how do you expect those to be integers?

Yeah, I realized that when I stepped away from my desk to run an errant.  Thanks for your patience.

closed 2020-02-26 by NateTG