openscad/openscad #3238: ERROR: CGAL error in CGALUtils::applyBinaryOperator UNKNOWN: CGAL ERROR: assertion violation! Expr: !Sphere_segment(ein->point(), lateral_sv_tgt[0]->point(), sphere_ray_src.sphere_circle()).is_long() File: /usr/include/CGAL/Convex_decomposition_3/Single_wall_creator3.h Line: 145

l29ah, 2020-02-28:
```
$fn=5;

use <Round-Anything/MinkowskiRound.scad>

minkowskiRound(0.5,2,1,[50,50,50])union(){//--example in the thiniverse thumbnail/main image
   cube([6,6,22]);
    rotate([30,45,10])cylinder(h=22,d=10);
}//--I rendered this out with a $fn=25 and it took more than 12 hours on my computer
```
as seen in https://github.com/Irev-Dev/Round-Anything results in
```
ERROR: CGAL error in CGALUtils::applyBinaryOperator UNKNOWN: CGAL ERROR: assertion violation! Expr: !Sphere_segment(ein->point(), lateral_sv_tgt[0]->point(), sphere_ray_src.sphere_circle()).is_long() File: /usr/include/CGAL/Convex_decomposition_3/Single_wall_creator3.h Line: 145 
ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation! Expr: e_below != SHalfedge_handle() File: /usr/include/CGAL/Nef_3/SNC_FM_decorator.h Line: 426 
```
OpenSCAD version 2020.02.17

l29ah, 2020-02-28:
Tried `minkowskiInsideRound` instead, got a segfault first, then
```
ERROR: CGAL error in CGALUtils::applyBinaryOperator difference: CGAL ERROR: assertion violation! Expr: itl != it->second.end() File: /usr/include/CGAL/Nef_3/SNC_external_structure.h Line: 1152
```

t-paul, 2020-02-28:
This is with `CGAL version, kernels: 4.11.3, Cartesian, Extended_cartesian, Epeck`