openscad/openscad #3242: Crash when "strangely" switching between apps

thomaskilian, 2020-03-07:
I'm using version 2019.12.20 (git 2a943a57) on MacOS Catalina and noticed that a "strange" switch between OpenSCAD and Cura (3.4.1) crashes OpenSCAD (and also sometimes Cura). I guess that both use the same GUI framework and would suspect that this is buggy under Catalina. Unfortunately I don't know how to provide valuable trace info. I can not constantly reproduce the crash, but it happens often when Alt-Tab between those apps.

t-paul, 2020-03-10:
Can you please check if there's any additional info in the logs shown via the "Console App"?

E.g. see https://blog.macsales.com/45586-how-to-use-the-mac-console-app-to-diagnose-a-crash/

thomaskilian, 2020-03-10:
Here are two logs where OpenSCAD and Cura crashed simultaneosly:

[Archiv.zip](https://github.com/openscad/openscad/files/4314941/Archiv.zip)

t-paul, 2020-03-12:
Thanks, it seems that crash happens across various applications. I found [QTBUG-52304](https://bugreports.qt.io/browse/QTBUG-52304) which is supposedly fixed in Qt5.9 but it's either not fixed or it's triggered here for a different reason.

kintel, 2024-11-28:
I assume this has since been fixed in Qt. Closing.

closed 2024-11-28 by kintel