openscad/openscad #3252: Jump to Definitions

Sharma-Hrishabh, 2020-03-18:
This enhancement aims to add a 'Jump to definition' feature inside the OpenScad editor. The possible suspects of jumps are from module/function calls to their declarations

Key-Points:
1: Handle multiple module/function declarations.

PRouzeau, 2020-03-26:
Yes, this is one of the most frequent move done in the editor. To do that, I'm using a trick since a few years:
On modules calls, there is no space between the module name and the opening parenthesis, while on the definition, there is a space between name and parenthesis, so I can jump to the def by selecting the call name and adding a space an opening parenthesis. Often, just adding the space is sufficient for discrimination.

But way too often I automatically type a space in the call, so I find regularly myself tracking those 'inappropriate' spaces.

MichaelAtOz, 2020-03-27:
I would also add variable assignments, and add 'Find definition' to the right-click menu, so you double click the name (function/module/variable) to select it, then right-click. 

> while on the definition, there is a space between name and parenthesis

There is no requirement for a space, I never use one. `module name() {}`

PRouzeau, 2020-03-27:
> There is no requirement for a space, I never use one. module name() {}

I know, this is just something I decided to use as a 'searching tool' and this proofed  very useful.