openscad/openscad #3254: 2D Projection error

gratefulfrog, 2020-03-24:
Hi,
I have discovered an odd behavior with 2D projection using the "cut=true" parameter.

This applies to version 2020.03.08.nightly as well as version 2015.03-2 .  I am running on Xubuntu 18.04.

The following code produces no result :

```
projection(true)
  translate([0,0,-5])
    import("Cam_First_Stage_2.stl");
```
But this code works fine, although it doesn't give me what I need...

```
projection(false)
  translate([0,0,-5])
    import("Cam_First_Stage_2.stl");
```

I hope this is of use to the great openscad community. An help would be appreciated!

[Cam_First_Stage_2.stl.zip](https://github.com/openscad/openscad/files/4374362/Cam_First_Stage_2.stl.zip)

MichaelAtOz, 2020-03-27:
The STL file is likely bad.
With `projection(true)` & F6 render I get a CGAL error (flush caches if you don't see it)
> ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation! Expr: e_below != SHalfedge_handle() File: /mxe/usr/x86_64-w64-mingw32.static.posix/include/CGAL/Nef_3/SNC_FM_decorator.h Line: 426

Also when `union()`ing it with a `cube()`.
That almost certainly means a bad STL, but I can't find a problem with my usual tools (Netfabb & Meshmixer).

Anyone else got other good tools?

gratefulfrog, 2020-03-27:
I have tried repairing the stl file with some tools too. But the "repaired" stl still produces nothing...

I get the following error message when cut=true, but not when false:

`ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation! Expr: e_below != SHalfedge_handle() File: /usr/include/CGAL/Nef_3/SNC_FM_decorator.h Line: 426 `

nophead, 2020-03-27:
IIRC I have had that problem when using complex geometry created by OpenSCAD rather than an STL. I think there is probably a bug somewhere but my examples we far too complex to post as a bug report, so I worked around it by extruding 2D instead of projecting 3D.

kintel, 2024-03-31:
Status as of 2024-03-24: Still broken both with CGAL and Manifold

kintel, 2024-04-13:
@JamesC1 Your issue is a different problem: You're trying to calculate a cutting plane at the extremity of an object, which will not be stable due to floating point uncertainties. Try translating your object a tiny bit in the positive z direction to make sure your object actually intersects the XY plane.

kintel, 2024-04-13:
Status as of 2024-04-12: Still broken both with CGAL, but this now works with Manifold after the fix in #5076

JamesC1, 2024-04-13:
@kintel, there were versions which exercised the cgal fault while being symmetric and large around the x-y plane.

Here's one with two different values for only one variable, one of which triggers the fault and the other doesn't.
https://pastebin.com/saqYBJeh

For each value of the parameter, I am showing you the pre- difference() object, the difference, and both projection(), with cut both true and false.

JamesC1, 2024-04-13:
Also exercises the fault with cubes:
https://pastebin.com/B3sEVtwP

kintel, 2024-04-13:
@JamesC1 Thanks - your new example demonstrates that this is still broken with CGAL, and works fine with Manifold.

JamesC1, 2024-04-13:
Thank you for confirming reproduction.  I am using openSCAD 2021.01 and started wondering if the envelope had changed.

difference() is not required in order to exercise the fault:
https://pastebin.com/6dDrvWYB

I hope to stop playing with this now : - )