openscad/openscad #3256: No auto-preview in completely empty files

stefanct, 2020-03-25:
I was trying to use openscad as an stl viewer to verify exports in a makefile by exploiting the empty file import idea from https://github.com/openscad/openscad/issues/3051#issuecomment-527343954 i.e. I tried executing `openscad -D 'import("some.stl")' empty.scad`.

This works beautifully if the file contains at least one character (e.g., `\n` or ` `). However, if the file is really empty, i.e. 0 bytes long, the preview is not triggered automatically on startup. After hitting F5 the view is updated correctly though. This is with Debian 10's OpenSCAD 2019.01~RC2-2.
This is independent from `Design->Automatic Reload and Preview` being enabled or disabled and is not related to the camera position.

t-paul, 2020-03-26:
The `-D` is meant for variable assignment, that is currently works as shown is a side effect of the implementation that is not really intended. I'd suggest using a variable for the filename and assign that via `-D`.

stefanct, 2020-03-26:
I am because I was expecting that. Still this behavior is funky enough to have it documented in an issue... feel free to close it as wontfix though. :)

t-paul, 2020-03-26:
I think there's a valid use case even without the `-D` when starting out with an empty file and giving that to both OpenSCAD and an external editor.

stefanct, 2020-03-26:
But that does not seem to be affected though? If I open up the empty file in openscad and change it externally then openscad is picking up the change and previews it correctly.

BTW: I have noticed another possible bug: should `--viewall` (and friends) work when opening an `.scad` file in GUI mode (instead of rendering a png)? It doesn't for me... new issue?

nophead, 2020-03-26:
I think OpenSCAD command line mode and GUI mode are considered separate. I am surprised -D affects GUI mode, but it does.  I can do ```openscad -Dx=1``` and then open a new file and type ```echo(x);```

t-paul, 2020-03-26:
Yes, I think that was specifically requested some time ago. I'm not sure it's a good idea.

MichaelAtOz, 2020-03-26:
@t-paul you often quote that [xfcd workflow](https://xkcd.com/1172/), if someone want to open their model with a custom default, who is to question the flexibility that it grants?

However there are probably a subset of options which are cmd-line only, it probably needs some thought?

MichaelAtOz, 2020-03-26:
> I can do openscad -Dx=1 and then open a new file and type echo(x);

Perhaps it is worth logging the -D to the GUI console as a reminder?

> it probably needs some thought?

For example animation could really use it.

stefanct, 2020-03-26:
> However there are probably a subset of options which are cmd-line only, it probably needs some thought?

I have quite some opinions to post... :) but since this is not related to my initial bug report I think we should open up a dedicated issue for that. Will do so shortly.

t-paul, 2020-03-26:
One problem is with multiple windows. Will `-D` affect only the first one? I have to check what the current behavior is...

kintel, 2025-04-08:
Closing as not planned. FYI: OpenSCAD supports input from stdin now, if you want to experiment without creating files.