openscad/openscad #3258: Tab bar hideable, not labeled

thehans, 2020-03-26:
I noticed this while clicking around the interface to see if there was any option related to my issue #3257

If you right click in the "label" area of the various panes: "Editor", "Console", or "Customizer", there is option to display/hide each, and also an unlabeled option, which seems to hide the tab bar when multiple documents are open.
[image]

I'm not sure if its even intentional to let user hide tabs like this or if its sort of an auto-populated list (I found I can still switch through documents with Ctrl-Tab when hidden, but it seems strange to hide them in the first place).
If intentional, it would be nice if this checkbox option could be labeled "Tabs". 
If unintended, then checkbox should be omitted.

MichaelAtOz, 2020-03-27:
Similar in Windows, but not in 2019.05.

Sharma-Hrishabh, 2020-04-02:
This context menu is generated by createPopupMenu() method of QMainWindow class automatically and covers Dockable Widgets.
[https://doc.qt.io/qt-5/qmainwindow.html#createPopupMenu](https://doc.qt.io/qt-5/qmainwindow.html#createPopupMenu) 
It can be overridden easily. So we can remove the unnamed option to hide/show tab toolbar or give it a name. 
I'm still not sure, whether this hide tabs action is required or not, so a little discussion will be helpful.

MichaelAtOz, 2020-04-04:
The Tab-bar hides when there is only one tab, which makes sense.
When there are multiple tabs you can move between them with Ctrl-Tab & Ctrl-Shift-Tab.
So the Tab-bar is not a required UI item, I suggest we name & keep it.

PR #3269 (2020-04-04): Fix unnamed tab toolbar title bug
Fixes #3258
closed 2020-04-04 by t-paul