openscad/openscad #3260: CLI options affecting GUI mode

stefanct, 2020-03-26:
Based on my observation in https://github.com/openscad/openscad/issues/3256#issuecomment-604175639_ and the ensued discussion it is clear that there are different views on how CLI options should or should not affect OpenSCAD's GUI mode.

@t-paul:
> I think OpenSCAD command line mode and GUI mode are considered separate. I am surprised -D affects GUI mode, but it does. I can do `openscad -Dx=1` and then open a new file and type `echo(x);`

@MichaelAtOz
> @t-paul you often quote that [xfcd workflow](https://xkcd.com/1172/), if someone want to open their model with a custom default, who is to question the flexibility that it grants?
> 
> However there are probably a subset of options which are cmd-line only, it probably needs some thought?

The following could be perceived as if I am cocky and demanding - it is just my personal view.
I don't understand the thinking behind having different options for GUI and CLI. I'd expect rather the contrary (and that's what I strive to implement when I work on something): There should be no action available on CLI that cannot also be made in the GUI (generally also the other way round would be great but that's usually not feasible - I cannot think of much in OpenSCAD that couldn't be done from CLI though since most of its interaction is text-based via the editor anyway). Therefore I think an application should do whatever I specify at the CLI no matter how it is displayed eventually.

Differentiating between different output modes unnecessarily makes the code more complicated in my experience and also requires additional documentation. The latter seems to be the main issue with OpenSCAD because AFAICT there is no clear indication which CLI arguments have an effect in GUI mode.

t-paul, 2020-03-26:
You are misquoting the first one. I do agree with what's said, but that comment was from @nophead.

stefanct, 2020-03-26:
Sorry, fixed.

nophead, 2020-03-26:
There is probably no technical reason why command line options couldn't be made affect GUI mode but they don't in general because it is a completely separate code flow. In the GUI the camera is controlled by the GUI and in command line mode via the options when generating PNGs.

I personally never launch GUI from the command line, I launch from the desktop, so there is no way to pass arguments. That is why I was surprised any have an effect. A lot of command line options suppress the GUI anyway and some would make little sense, for example outputting dependencies.

Some command line options are to replace registry settings. For example ```--hardwarnings``` is a preference in GUI mode and a command line option when the GUI isn't launched. Similarly ```--coloscheme```. What would they do in GUI mode when passed from the command line? Would they override the registry and would they write back to it?

donbright, 2020-03-26:
well what had happened was..a long long time ago,  the "command line mode" was rather sparse.  Marius wrote the regression tests and they did not use the same front end code as the GUI program code. They didn't really use much, if any QT (the gui framework library). The tests had their own 'front ends' to the openscad "engine". Some of these tests would output a PNG file to disk, for self-testing purposes. 

Then along came Some User. Some User wanted a command line version of openscad, to be able to output PNG files, 2d images of the rendered objects. 

So some magical elves went into the Test Suite code, and pulled out a bunch of chunks of it, and integrated them into the GUI code, so then the GUI program could dump PNGs to a file. 

Then everyone, who had a knack for it, lived happily ever after.