openscad/openscad #3261: New features that should be added.

VendicarKahn, 2020-03-26:


VendicarKahn, 2020-03-26:
New features that should be added.

1: Return a child's Bounding Box.

2: Matrix multiplication of mesh with the matrix coefficients being functions of the mesh coordinates x,y,z should be provided.

I.E. matrixmul {[[ cos(&z), -sin)&z),0,0],
                        [sin(&z),cos(&z),0,0],
                        [0,0,1,0],[0,0,0,1]])
children();

Thereby allowing object and surfaces to be wrapped around cylinders, and spheres.

3: A function for returning access to mesh data should be provided.

4: Surface features should support labeling.

5: Once labeled a function to return surface properties should be provided - mesh, normal, etc.

6: A method should be provided to search for a labeled mesh object and allow it to be queried to return it's orientation, and extent, and point map.

t-paul, 2020-03-26:
I'm sorry, please don't create issues with multiple completely unrelated topics. Also most of those sound like they need much more detailed description.

closed 2020-03-26 by t-paul

MichaelAtOz, 2020-03-27:
@VendicarKahn ...and conceptual discussion can best be done on the [forum](http://forum.openscad.org/) or [mailing-list](url) (see first pinned post on the forum for details).

t-paul, 2020-03-27:
Hi @VendicarKahn closing the issue was not meant to discurage posting issues in general. It's just not possible to handle multiple topics together. Things might be implemented by different people, some may need to wait quite some time if nobody has time to work on them. That results in an ever more complicated status of the issue which can't be handled in a small amount of spare time.

So maybe start with that matrix calculation as separate issue? Do you have some examples of how to use it for specific modelling effects? We also have now function arguments (lambdas) that might factor into this discussion.

VendicarKahn, 2020-04-05:
The point is, you don't want to implement simple matrix calculations because they aren't capable of producing non-linear transformations, non-linear projections, etc.  There is a need for doing these things so that wraps and curves can be better implemented without the need to slice up a model and distort the individual slices.  Doing this is slow to calculate and will leave the final model unnecessarily complex as overlaps and jagged edges will consume more processing time in the final model. With a non-linear operation the mesh itself is modified so that the body itself remains intact - no holes - and with none of the jaggies that necessarily result from a slicing and splicing method. To this end the most general solution would be to run the mesh through a user defined filter, taking a 2 or 3 point from the mesh and returning a new 2 or 3 point which is then re-inserted into the mesh. The filter transform syntax is not particularly important, but as a model I imagine... Function(x,y,z) - {} TransformMesh(FunctionName); In addition some meshes will require increased resolution along edges that will ultimately become curves in order to approximate the curve. The solution here is to extend the mesh through subdivision along each edge. I would recommend subdivision based on spanning distance between two end points. So you process the edge map.  If and edge is longer than some limit, insert a new point at the midpoint and replace the edge with two edges that are co-liner. Following this operation a non-linear transformation has the potential of more realistic curves after a non-linear transform is performed. Another feature not mentioned earlier would also be useful.  Maintaining the extent of an internal bounding cube of the implied object currently generated.

VendicarKahn, 2020-04-05:
"Do you have some examples of how to use it for specific modelling effects?" Non linear transformations are used in your own software for performing paralax projection conversion in the model display window. The are a needed component for performing a variety of warps in the transformation of surface and surface features.  For example wrapping flat text over a surface, or transforming other objects - say a grill or a button surface texture to the curved surface of a button, or an embossed emblem onto the curved surface of a model above a button. Also the generation of various map projections, curved mirrors, etc. None of these things can be performed by simple matrix transformations that implement the matrix as a set of constants. For these things you need x = a * f(x') + b * f(y') + c * f(z')

MichaelAtOz, 2020-04-06:
[I edited the last two posts on Github, added paragraphs, easier to read]