openscad/openscad #3264: parent_module() wrong return

RonaldoCMP, 2020-03-29:
That is what I get with just the test code loaded to the editor:

[image]

But when other codes are loaded a different result comes out:

[image]

show_quad_mesh() is a function defined in new_CP.scad

t-paul, 2020-03-29:
Please add the text of the scripts in addition.

RonaldoCMP, 2020-03-29:
>
> Please add the text of the scripts in addition.
>
>
Do you mean the code of newCP.scad? That is a somewhat big code I am
working on. But even after I close it the the parent_module code still
returns the same string "show_quad_mesh".

>
>

t-paul, 2020-03-29:
Everything related, specifically including the code from the images. It seems like reasonably small test case. That might increase the chances someone tries to fix the issue.

nophead, 2020-03-29:
Isn't parent_module(2) out of bounds when there is just one parent?

t-paul, 2020-03-29:
Yes, I'm wondering about that too. The code seems to have a valid bounds check, but maybe there's some wrong data in that stack of module names.

RonaldoCMP, 2020-03-29:
It should and do it ... sometimes.

In a fresh window, run:
module top() {
   children();
 }

x = 0;
top() echo(parent_module(x)); 

It should corretly echo "top". Now, initialize x with 3. It will warn that x is greater than the number of modules on the stack. Then reset x to 1. then 2, then 3. It will echo "top" in the first two cases and issue a warn to the  last one.  Set x to 4 then reset to 3 again. It  will echo "top" once more.

It seems that it retains some memory from previous runs.

I am unable to reproduce the first case a have reported before.

I using version 2019.10.25.ci3851 (git feee8f02)

t-paul, 2020-03-29:
Oh, thanks, that's a good hint. So it probably does not clean out the module name stack at all. This would only be visible when trying to access out of bounds index values.

t-paul, 2020-04-11:
I can' reproduce that, tested
- Fresh debug build from master (Debian/Testing)
- Fresh release build from master (Debian/Testing)
- 2019.10.25.ci3851 downloaded from file server via `wine`

Alfriadox, 2020-04-26:
I am also having this issue. OpenSCAD 2019.05 for Windows. 
```
/*
Nia Calia-Bogan
April 25th, 2020
OpenSCAD 2019.05

Utilities for tracing through parents of a given.
*/

// Module that immediately calls children, without any operation.
module Nop() {children();echo(_list_parents());}

// Returns list of parent modules. 
// Currently not working.
function _list_parents() = 
    [for (i = [0:$parent_modules-1]) parent_module(i)];

module ListChildren() {
    children();
}

ListChildren() Nop();
```
output:
`ECHO: ["Nop", "ListChildren", "AssertIsList"]`

Alfriadox, 2020-04-26:
For context, there are several other files in the same directory (none are referenced by this file). One of those files contains a module called AssertIsList. That file happens to be open in another window.

Alfriadox, 2020-04-26:
If all the windows are closed though, and then this file is reopened, the bug disappears.

Alfriadox, 2020-04-26:
I think somewhere stuff just needs to get flushed or reloaded or something?

t-paul, 2020-04-26:
Yes, see my suspicion above, it's just hard to do a fix if it's not reproducible on the dev machine. I'll try again with some combinations of multiple windows and tabs.

t-paul, 2020-05-06:
The [forum post](http://forum.openscad.org/parent-modules-increases-each-time-the-recursion-limit-is-broken-in-2019-5-0-td29050.html) seems to have given the important hint that it's happening on compile errors. @RonaldoCMP do you have "Stop on first Warning" active?

RonaldoCMP, 2020-05-06:
Usually yes but in my reported case there was no warning.

t-paul, 2020-05-06:
So far I could only reproduce after an assert or compile error happened (potentially in a different tab) and that should be fixed in:

[Windows 32bit](https://4978-1049088-gh.circle-artifacts.com/0/32-bit/OpenSCAD-2020.05.06.ci4978-x86-32_fix-parent-module-stack.zip)
[Windows 64bit](https://4979-1049088-gh.circle-artifacts.com/0/64-bit/OpenSCAD-2020.05.06.ci4979-x86-64_fix-parent-module-stack.zip)
[Linux AppImage](https://4977-1049088-gh.circle-artifacts.com/0/64-bit/OpenSCAD-2020.05.06.ai4977-_fix-parent-module-stackd7da70a-x86_64.AppImage)

RonaldoCMP, 2020-05-09:
Yes! It worked nice on my Windows 64 system.

commit 81f3777c (2020-05-22): Merge pull request #3303 from openscad/fix-parent-module-stack

Fix parent module handling (fixes #3264).
closed 2020-05-22 by t-paul