openscad/openscad #3271: not seeming to use `--autocenter --viewall" command line arguments

timmit99, 2020-04-06:
I have 30+ STL's that i need to generate images for however that are not centered on 00 as they were exported relative to their assembly's origin and are all over the XYZ space. If i MANUALLY load them up i can hit "View All" and it centers but running "openscad file.stl --viewall -o out.png" yields a blank yellow image.

Directory is:
Images (folder)
STLs (folder)
model.scad
Image_Generation.py

```python
# Image_Generation.py
import os

directory = r'STLs'
accent = "[a]"
for entry in os.scandir(directory):
    if (entry.path.lower().endswith(".stl") and entry.is_file()):
        print(entry.path)
        fileName = entry.path[len(directory)+1:]
        justName = fileName[0:len(fileName)-4]
        if (fileName.startswith(accent)):
            color = "accent"
        else:
            color = "main"
        print(fileName,"   ",justName)

        command = "openscad model.scad -D model=\"" +justName+ "\" -D type=\"" +color+ "\" -o Images/" + justName + ".png --viewall  --colorscheme DeepOcean "
        print(command)
        os.system(command)
```
and

```openSCAD
model="temp";
type = "main";
filename=str("\STLs\\",model,".STL");
import(filename, center=true);
color("red")
import(filename, center=false);
```
If anyone can help me out... I've tried just:
`openscad test.stl -o Images/test.png --viewall  --colorscheme DeepOcean` in command line but still no success

MichaelAtOz, 2020-04-06:
With 2019.05 on Windows &
```
translate([-95,0]) cube(3);
translate([-90,0]) cube(3);
```
and 
```
openscad viewall.scad -o viewall-va.png

```
I get an image with the two cubes really close up, centred in the png.
Adding `--viewall` produced identical output.
A changing one of the minus translates to positive, give a similar full image showing both cubes further zoomed out.

I exported the two minus cubes as .STL, then changed the .SAD to `import("viewall.stl);`, then
`openscad viewall.scad -o viewall-cs-va.png --colorscheme DeepOcean --viewall`
got the same close up cubes, and the same without `--viewall`

So I don't know what you are doing to not get centred output...

MichaelAtOz, 2020-04-06:
and `--autocenter' without `--viewall` similarly same output.

nophead, 2020-04-06:
```--viewall``` and ```--autocenter``` are both needed.

```autocenter``` points the camera at the centre of the object. ```viewall``` sets the camera distance to ensure the bounding sphere fits in the height of the image. That is a bit pessimistic in most cases so I make the image oversized and then use Imakemagick to trim the excess border and resample it to a lower resolution. That gives me anti-aliasing. For example I start with a 4096 x 4096 image to make a 280x280 one of each STL in my build instructions.

[image]

nophead, 2020-04-06:
This is what my frame work generates for the two cubes.

[image]

t-paul, 2020-04-06:
@timmit99 please post a specific example STL + full command line that is not working.

closed 2020-05-09 by t-paul

spuder, 2021-12-26:
I also am seeing that `--viewall` and `--autocenter` appear to not work (Running 2021.01)

```
find ~+ -type f -name "*.stl" -print0 | while read -d '' -r file; do 
	echo "Reading $file"
	/Applications/OpenSCAD.app/Contents/MacOS/OpenSCAD /dev/null -D "import(\"$file\");" -o "$file.png" --imgsize=600,600 --colorscheme "Tomorrow Night" --viewall --autocenter
done
```

The md5sum of files with and without `--viewall --autocenter` is identical. 

I've also tried using the options when generating a gif, and openscad also produces the same result regardless if `--viewall --autocenter` is defined or not

```
find ~+ -type f -name "*.stl" -print0 | while read -d '' -r file; do 
    echo "Reading $file"
    MYTMPDIR="$(mktemp -d)"
    trap 'rm -rf -- "$MYTMPDIR"' EXIT
    echo "Creating temp directory ${MYTMPDIR}"
    /Applications/OpenSCAD.app/Contents/MacOS/OpenSCAD /dev/null -D '$vpd = 100;' -D '$vpr = [60, 0, 360 * $t];' -o "${MYTMPDIR}/foo.png"  -D "import(\"$file\");" --imgsize=300,300 --animate 60 --colorscheme "Tomorrow Night" --viewall --autocenter
    yes | ffmpeg \
        -framerate 12 \
        -pattern_type glob \
        -i "$MYTMPDIR/*.png" \
        -r 24 \
        -vf scale=512:-1 \
        "${file}.gif" \
        ;
    rm -rf -- "$MYTMPDIR"
done
```

rcolyer, 2021-12-26:
Since @nophead's 2018 commit d57e6e990aad15a646ba9961bebdfda2e3937214 the --viewall and --autocenter options are defaulted to on whenever the --camera parameter is not specified.  Actually displaying what was generated seems to me to have been a reasonable change in default behavior.  Their only remaining role is to override values passed for camera, and autocenter is only processed if viewall is also enabled.  So it's not that they don't work, it's that there is a very limited case where they are not working.  I suspect this is NotABug.

Three different cases:
```openscad /dev/null -D "translate([20,0,0]) cube(10);" -o test_None.png```
[image]
```openscad /dev/null --camera 30,0,0,70,0,20,50 -D "translate([20,0,0]) cube(10);" -o test_C.png```
[image]
```openscad /dev/null --camera 30,0,0,70,0,20,50 --viewall --autocenter -D "translate([20,0,0]) cube(10);" -o test_CVA.png```
[image]