openscad/openscad #3272: len(num) = error

VendicarKahn, 2020-04-06:
len(num) should return 1, not generate an error.

if possible num[0] and num.x should also resolve to num.

t-paul, 2020-04-06:
If num is a number, then no. A number has no length and you can't index it either.

VendicarKahn, 2020-04-06:
A number is a vector of length 1. it is also a matrix of dimension [1]. These are mathematical facts. Are you unaware of them?

t-paul, 2020-04-06:
We are not talking math here, but programming language which may have some differences and also needs to consider backward compatibility. I'm leaving this open for discussion if someone brings additional arguments to the table.

Is there any other programming language which behaves that way?

nophead, 2020-04-06:
In maths scalars and vectors are different things.

[1] is a vector of length 1. 
1 is a scalar, for which length has no meaning.

adrianVmariano, 2020-04-08:
In math, a vector space is a set which has addition and scalar multiplication.  The set of real numbers is a vector space of dimension 1.  I agree with VendicarKahn on this one.  Real numbers can be appropriately regarded as vectors with dimension 1.

The MATLAB programming language does not distinguish between scalars and length 1 vectors (or 1x1 matrices).   I do fairly routinely write code to accept a scalar or a vector, so having them treated the same would simplify some of my code.  However, this would be a pretty big change and the full ramifications are not obvious to me.  I'd be concerned about that.  It doesn't seem like this change would be beneficial enough to devote developer resources to it, or deal with the possible compatibility issues.

donbright, 2020-04-08:
a group of engineers were trying to determine the height of a flag pole. 

they tried to calculate using the sun, shadows, angles, laser finders, and so forth, but the youngest , newest engineer took the pole out of it's cradle, layed it flat on the ground and measured it with a tape measure

the older engineers all laughed and said "well that was pointless, we wanted the height and she measured the length!"

VendicarKahn, 2020-04-11:
"a scalar is a vector of length 1" Matrix Algebra: Theory, Computations and Applications in Statisticshttps://books.google.ca/books?id=HdQ5DwAAQBAJ&pg=PA544&lpg=PA544&dq=%22a+scalar+is+a+vector+of+length+1%22&source=bl&ots=Z9_pQ_B8hL&sig=ACfU3U3Ks2swfCyFBPbY6xmLrS582Hy5UA&hl=en&sa=X&ved=2ahUKEwiKu-v0pN_oAhURCs0KHdPCCXUQ6AEwAXoECAsQKQ#v=onepage&q=%22a%20scalar%20is%20a%20vector%20of%20length%201%22&f=false

VendicarKahn, 2020-04-11:
Are there any other programming languages where a*b results in a dot product if a and b are vectors? What you need to do is return len(scalar) = 1 so that len(num) is the same as len([num]) and you resolve num[0] = num

MichaelAtOz, 2020-04-11:
Unfortunately the OpenSCAD language ecosystem pays a lot of attention to backwards compatibility.
`len(1)` does not return an error, it returns `undef`.
Changing len() in that way will break backward compatibility, `len(num) == undef` is used as a type test.

@VendicarKahn you are able to write your own len function and have it return what ever you want.

```
s=1;
v=[1];
echo("len(),",len(1),len(v));

function lenv(q) =  is_undef(q) ? undef
                    : is_num(q) ? 1
                      : is_list(q) ? len(q)
                        : is_string(q) ? len(q)
                          : undef;// bool/function
                          
echo("lenv(),",lenv(s),lenv(1));
echo("lenv(),",lenv(v),lenv([1]));
```

VendicarKahn, 2020-04-11:
len(num) == undef is used as a type test. That would be foolish as there is a function is_num is provided for that purpose. Using len(num) == undef is at best relying on a side effect and should be considered as unreliable unsupported as all side effects should be.

MichaelAtOz, 2020-04-11:
'is_num()` et al  was only introduced in 2019.05, `len(num) == undef` was the only way for the many years OpenSCAD has been around.

MichaelAtOz, 2020-04-11:
...and it is not a side effect. It is documented expected behaviour of the OpenSCAD language. The language has 'Number' & 'Vector' which are different. You want to change the fundamentals of the language.

VendicarKahn, 2020-04-12:
A number - a scalar - is a single element vector.  That is how numbers are defined.

closed 2020-04-12 by t-paul