openscad/openscad #3273: Preferences.cc:(.text+0x67): undefined reference to `InitConfigurator::InitConfigurator()'

kunyang-yk, 2020-04-07:
[ 97%] Linking CXX executable openscad
/usr/bin/ld: CMakeFiles/OpenSCAD.dir/src/Preferences.cc.o: in function `Preferences::Preferences(QWidget*)':
Preferences.cc:(.text+0x67): undefined reference to `InitConfigurator::InitConfigurator()'
/usr/bin/ld: CMakeFiles/OpenSCAD.dir/src/Preferences.cc.o: in function `Preferences::init()':
Preferences.cc:(.text+0x1323): undefined reference to `InitConfigurator::initComboBox(QComboBox*, Settings::SettingsEntry const&)'
/usr/bin/ld: Preferences.cc:(.text+0x134a): undefined reference to `InitConfigurator::initComboBox(QComboBox*, Settings::SettingsEntry const&)'
/usr/bin/ld: Preferences.cc:(.text+0x1371): undefined reference to `InitConfigurator::initComboBox(QComboBox*, Settings::SettingsEntry const&)'
/usr/bin/ld: Preferences.cc:(.text+0x1398): undefined reference to `InitConfigurator::initComboBox(QComboBox*, Settings::SettingsEntry const&)'
/usr/bin/ld: Preferences.cc:(.text+0x13bf): undefined reference to `InitConfigurator::initComboBox(QComboBox*, Settings::SettingsEntry const&)'
/usr/bin/ld: CMakeFiles/OpenSCAD.dir/src/Preferences.cc.o:Preferences.cc:(.text+0x13e6): more undefined references to `InitConfigurator::initComboBox(QComboBox*, Settings::SettingsEntry const&)' follow
/usr/bin/ld: CMakeFiles/OpenSCAD.dir/src/Preferences.cc.o: in function `Preferences::init()':
Preferences.cc:(.text+0x1447): undefined reference to `InitConfigurator::initSpinBoxRange(BlockSignals<QSpinBox*> const&, Settings::SettingsEntry const&)'
/usr/bin/ld: Preferences.cc:(.text+0x148d): undefined reference to `InitConfigurator::initSpinBoxRange(BlockSignals<QSpinBox*> const&, Settings::SettingsEntry const&)'
/usr/bin/ld: Preferences.cc:(.text+0x14d3): undefined reference to `InitConfigurator::initSpinBoxRange(BlockSignals<QSpinBox*> const&, Settings::SettingsEntry const&)'
/usr/bin/ld: Preferences.cc:(.text+0x1519): undefined reference to `InitConfigurator::initSpinBoxRange(BlockSignals<QSpinBox*> const&, Settings::SettingsEntry const&)'
/usr/bin/ld: Preferences.cc:(.text+0x154c): undefined reference to `InitConfigurator::initComboBox(QComboBox*, Settings::SettingsEntry const&)'
/usr/bin/ld: Preferences.cc:(.text+0x1573): undefined reference to `InitConfigurator::initComboBox(QComboBox*, Settings::SettingsEntry const&)'
/usr/bin/ld: CMakeFiles/OpenSCAD.dir/src/Preferences.cc.o: in function `Preferences::updateGUI()':
Preferences.cc:(.text+0x7a3d): undefined reference to `InitConfigurator::updateComboBox(BlockSignals<QComboBox*> const&, Settings::SettingsEntry const&)'
/usr/bin/ld: Preferences.cc:(.text+0x7a83): undefined reference to `InitConfigurator::updateComboBox(BlockSignals<QComboBox*> const&, Settings::SettingsEntry const&)'
/usr/bin/ld: Preferences.cc:(.text+0x7ac9): undefined reference to `InitConfigurator::updateComboBox(BlockSignals<QComboBox*> const&, Settings::SettingsEntry const&)'
/usr/bin/ld: Preferences.cc:(.text+0x7b0f): undefined reference to `InitConfigurator::updateComboBox(BlockSignals<QComboBox*> const&, Settings::SettingsEntry const&)'
/usr/bin/ld: Preferences.cc:(.text+0x7b55): undefined reference to `InitConfigurator::updateComboBox(BlockSignals<QComboBox*> const&, Settings::SettingsEntry const&)'
/usr/bin/ld: CMakeFiles/OpenSCAD.dir/src/Preferences.cc.o:Preferences.cc:(.text+0x7b9b): more undefined references to `InitConfigurator::updateComboBox(BlockSignals<QComboBox*> const&, Settings::SettingsEntry const&)' follow
/usr/bin/ld: CMakeFiles/OpenSCAD.dir/src/Preferences.cc.o: in function `Preferences::updateGUI()':
Preferences.cc:(.text+0x7c27): undefined reference to `InitConfigurator::initSpinBoxDouble(BlockSignals<QSpinBox*> const&, Settings::SettingsEntry const&)'
/usr/bin/ld: Preferences.cc:(.text+0x7c6d): undefined reference to `InitConfigurator::initSpinBoxDouble(BlockSignals<QSpinBox*> const&, Settings::SettingsEntry const&)'
/usr/bin/ld: Preferences.cc:(.text+0x7cb3): undefined reference to `InitConfigurator::initSpinBoxDouble(BlockSignals<QSpinBox*> const&, Settings::SettingsEntry const&)'
/usr/bin/ld: Preferences.cc:(.text+0x7cf9): undefined reference to `InitConfigurator::initSpinBoxDouble(BlockSignals<QSpinBox*> const&, Settings::SettingsEntry const&)'
/usr/bin/ld: Preferences.cc:(.text+0x7d3f): undefined reference to `InitConfigurator::initCheckBox(BlockSignals<QCheckBox*> const&, Settings::SettingsEntry const&)'
/usr/bin/ld: Preferences.cc:(.text+0x7d85): undefined reference to `InitConfigurator::initCheckBox(BlockSignals<QCheckBox*> const&, Settings::SettingsEntry const&)'
/usr/bin/ld: Preferences.cc:(.text+0x7dcb): undefined reference to `InitConfigurator::initCheckBox(BlockSignals<QCheckBox*> const&, Settings::SettingsEntry const&)'
/usr/bin/ld: Preferences.cc:(.text+0x7e11): undefined reference to `InitConfigurator::initCheckBox(BlockSignals<QCheckBox*> const&, Settings::SettingsEntry const&)'
/usr/bin/ld: Preferences.cc:(.text+0x7e57): undefined reference to `InitConfigurator::initCheckBox(BlockSignals<QCheckBox*> const&, Settings::SettingsEntry const&)'
/usr/bin/ld: CMakeFiles/OpenSCAD.dir/src/Preferences.cc.o:Preferences.cc:(.text+0x7e9d): more undefined references to `InitConfigurator::initCheckBox(BlockSignals<QCheckBox*> const&, Settings::SettingsEntry const&)' follow
/usr/bin/ld: CMakeFiles/OpenSCAD.dir/src/Preferences.cc.o: in function `Preferences::updateGUI()':
Preferences.cc:(.text+0x81a8): undefined reference to `InitConfigurator::updateComboBox(BlockSignals<QComboBox*> const&, Settings::SettingsEntry const&)'
/usr/bin/ld: Preferences.cc:(.text+0x81ee): undefined reference to `InitConfigurator::updateComboBox(BlockSignals<QComboBox*> const&, Settings::SettingsEntry const&)'
/usr/bin/ld: Preferences.cc:(.text+0x8234): undefined reference to `InitConfigurator::updateComboBox(BlockSignals<QComboBox*> const&, Settings::SettingsEntry const&)'
/usr/bin/ld: CMakeFiles/OpenSCAD.dir/src/input/AxisConfigWidget.cc.o: in function `AxisConfigWidget::AxisConfigWidget(QWidget*)':
AxisConfigWidget.cc:(.text+0x51): undefined reference to `InitConfigurator::InitConfigurator()'
/usr/bin/ld: CMakeFiles/OpenSCAD.dir/src/input/AxisConfigWidget.cc.o: in function `AxisConfigWidget::init()':
AxisConfigWidget.cc:(.text+0x879): undefined reference to `InitConfigurator::initComboBox(QComboBox*, Settings::SettingsEntry const&)'
/usr/bin/ld: AxisConfigWidget.cc:(.text+0x8a0): undefined reference to `InitConfigurator::initComboBox(QComboBox*, Settings::SettingsEntry const&)'
/usr/bin/ld: AxisConfigWidget.cc:(.text+0x8c7): undefined reference to `InitConfigurator::initComboBox(QComboBox*, Settings::SettingsEntry const&)'
/usr/bin/ld: AxisConfigWidget.cc:(.text+0x8ee): undefined reference to `InitConfigurator::initComboBox(QComboBox*, Settings::SettingsEntry const&)'
/usr/bin/ld: AxisConfigWidget.cc:(.text+0x915): undefined reference to `InitConfigurator::initComboBox(QComboBox*, Settings::SettingsEntry const&)'
/usr/bin/ld: CMakeFiles/OpenSCAD.dir/src/input/AxisConfigWidget.cc.o:AxisConfigWidget.cc:(.text+0x93c): more undefined references to `InitConfigurator::initComboBox(QComboBox*, Settings::SettingsEntry const&)' follow
/usr/bin/ld: CMakeFiles/OpenSCAD.dir/src/input/AxisConfigWidget.cc.o: in function `AxisConfigWidget::init()':
AxisConfigWidget.cc:(.text+0xca4): undefined reference to `InitConfigurator::initCheckBox(BlockSignals<QCheckBox*> const&, Settings::SettingsEntry const&)'
/usr/bin/ld: AxisConfigWidget.cc:(.text+0xcf3): undefined reference to `InitConfigurator::initCheckBox(BlockSignals<QCheckBox*> const&, Settings::SettingsEntry const&)'
/usr/bin/ld: AxisConfigWidget.cc:(.text+0xd42): undefined reference to `InitConfigurator::initCheckBox(BlockSignals<QCheckBox*> const&, Settings::SettingsEntry const&)'
/usr/bin/ld: AxisConfigWidget.cc:(.text+0xd91): undefined reference to `InitConfigurator::initCheckBox(BlockSignals<QCheckBox*> const&, Settings::SettingsEntry const&)'
/usr/bin/ld: AxisConfigWidget.cc:(.text+0xde0): undefined reference to `InitConfigurator::initCheckBox(BlockSignals<QCheckBox*> const&, Settings::SettingsEntry const&)'
/usr/bin/ld: AxisConfigWidget.cc:(.text+0x1074): undefined reference to `InitConfigurator::initDoubleSpinBox(QDoubleSpinBox*, Settings::SettingsEntry const&)'
/usr/bin/ld: AxisConfigWidget.cc:(.text+0x11a7): undefined reference to `InitConfigurator::initDoubleSpinBox(QDoubleSpinBox*, Settings::SettingsEntry const&)'
[... cut ...]
/usr/bin/ld: AxisConfigWidget.cc:(.text+0x120d): undefined reference to `InitConfigurator::initDoubleSpinBox(QDoubleSpinBox*, Settings::SettingsEntry const&)'
/usr/bin/ld: AxisConfigWidget.cc:(.text+0x1234): undefined reference to `InitConfigurator::initDoubleSpinBox(QDoubleSpinBox*, Settings::SettingsEntry const&)'
/usr/bin/ld: CMakeFiles/OpenSCAD.dir/src/input/AxisConfigWidget.cc.o:AxisConfigWidget.cc:(.text+0x125b): more undefined references to `InitConfigurator::initDoubleSpinBox(QDoubleSpinBox*, Settings::SettingsEntry const&)' follow
collect2: error: ld returned 1 exit status

t-paul, 2020-04-07:
Please add some environment information. Which code version? Which distro (as that looks like it might be Linux)? What build command?

Are you using `qmake`?

kunyang-yk, 2020-04-07:
cmake build.

Add in CMakeLists.txt file
### src/initConfig.cc###

t-paul, 2020-04-07:
`cmake` is not officially supported yet (#2019 / #2518). It hopefully will be at some point, but right now, the build system is still `qmake`.

closed 2020-04-07 by t-paul

t-paul, 2020-04-08:
I've added a `cmake` based test run that should guard against too many future issues with the `cmake` setup. But for the time beeing `qmake` remains the offical build system.