openscad/openscad #3275: imported STL renders in overview, but disappears when pressing F6

gnthibault, 2020-04-10:
This is just utterly frustrating to realize all the work is useless although it was previously rendering properly.

To reproduce just run the scad file:

[dustcap2.zip](https://github.com/openscad/openscad/files/4461063/dustcap2.zip)

gnthibault, 2020-04-10:
When ran in command line, I get:

Error !!!!!!!!!!!!!!!!!!!!!!!
ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation!
Expr: pe_prev->is_border() || !internal::Plane_constructor<Plane>::get_plane(pe_prev->facet(),pe_prev->facet()->plane()).is_degenerate()
File: /usr/include/CGAL/Nef_3/polyhedron_3_to_nef_3.h
Line: 251

gnthibault, 2020-04-10:
Ok whatever I try to do with the imported stl (union, difference) the rendering just fails.
Importing the stl and redering with a simple scaling works as expected

nophead, 2020-04-10:
In that case your STL is not manifold. There has to no or self intersections or degenerate triangles for CGAL to accept it.

gnthibault, 2020-04-10:
The stl is perfectly valid I think, tested with: https://github.com/openscad/openscad/blob/master/tests/validatestl.py

nophead, 2020-04-10:
That is far too simple a test to detect self intersections.

gnthibault, 2020-04-10:
OK any workaround ?

gnthibault, 2020-04-10:
Ok freecad is really an awesome tool, looks like it managed to fix the mesh.

closed 2020-04-10 by gnthibault