openscad/openscad #3277: 2d Projection created by multmatrix can not be extruded.

VendicarKahn, 2020-04-11:
Doing a manual projection with a user defined matrix [[1,0,0,0],[0,1,0,0],[0,0,0,0],[0,0,0,1]]
results in a 2d projection onto the xy plane.

Requesting a linear extrude of this 2d object residing on the xy plane results in an error extruding a 3d object when no 3d object exists.

nophead, 2020-04-11:
Un-surprising as you haven't made a 2D object, you have just squashed a 3D one, which will no longer be manifold due to lots of self intersections.

t-paul, 2020-04-11:
As @nophead already said, a matrix tranformation does not convert 3D to 2D. For that there's a special module [`projection()`](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Using_the_2D_Subsystem#3D_to_2D_Projection).

closed 2020-04-11 by t-paul

VendicarKahn, 2020-04-12:
So you are saying that there are two kinds of 2d objects in openscad. The first has zero dimensions along one axis and the second has zero directions along one axis and has special magical properties. Unacceptable.

nophead, 2020-04-12:
No there are only 2D objects with XY coordinates and 3D objects with X, Y, Z coordinates.

If a 3D object has all its Z values 0 it is an invalid 3D object as 3D objects represent 3D solids and must have a finite thickness.

t-paul, 2020-04-12:
OpenSCAD is mesh based. If you squish a (default aligned) cube with 6 polygons to zero height, you still end up with a degenerated 3d object having 4 zero area polygons and 2 with the same coordinates but different normal vector.

VendicarKahn, 2020-04-12:
Manifolds are only defined in 3 or more dimensions.  So, ya.  Using a matrix to project a 3d object onto a 2d surface necessarily means not having a manifold. The projection function also does not return a manifold since it returns a 2d object. There is no such thing as a 2-manifold. "Lots of self intersections" -  That is your problem, not mine. If you provide a multmatrix function you should support any matrix, not just an unspecified subset of them.

VendicarKahn, 2020-04-12:
If you provide a multmatrix function then you should support all matricies, not just an unspecified subset of them. What other matrix operations don't you support.

VendicarKahn, 2020-04-12:
Maybe you should actually write that down somewhere.  You know, like in the documentation.  Especially in the Multmatrix section of the documentation to make it clear that this program is incapable of managing   all matrix operations. LOL It's half baked.

t-paul, 2020-04-12:
Ok, I think that is enough.

nophead, 2020-04-12:
The matrix operations are fine but just like polyhedron they are GIGO. You can make an infinite number of 3D objects that CGAL will not accept because they are not manifolds. 

OpenSCAD provides projection() to turn a 3D object into a 2D object. It doesn't simply set all the Z coordinates to 0. It creates a new polygon outline like a shadow and removes all the internal edges that would collapse on top of each making a polygon with lots of self intersections.

On the other hand multmatrix does a matrix multiplication on all the vertices. It doesn't add or remove any of them or change the edges.