openscad/openscad #3278: Projection matrix

sbridger, 2020-04-12:
Some years ago I was trying to use a perspective projection matrix transform (multimatrix) to do conical extrusion
i.e. set the transform matrix to be a projection up Z. If you extrude a circle, you will get a cone. I have often wanted to do a conical extrusion, e.g. to get draught.

I thought I had it working at one point, but a while later when I went back (probably after an update) I was unable to get this to work anymore.

Is there a reason why these transforms don't, and did it change at some point, or it just a side effect of the very cheap gin I drink?

t-paul, 2020-04-12:
Matrix transformation can't create points and 2D shapes currently have no Z, so I doubt that ever worked. For questions please prefer the forum/mailing list.
If you extrude a circle into a cone, the way would be `linear_extrude(..., scale = 0.5);`.

closed 2020-04-12 by t-paul

sbridger, 2020-04-12:
Thanks for scale.

I don't really understand what you mean by
> Matrix transformation can't create points and 2D shapes currently have no Z

 MM works on the 3D extruded objects space. (I assume that like postscript you keep a single "current transform matrix" ) 
For example when we shear:
```
m = [ [ 1  , 0  , 0  , 0   ],
      [ 0  , 1  , 0.7, 0   ],  // The "0.7" is the skew value; pushed along the y axis as z changes.
      [ 0  , 0  , 1  , 0   ],
      [ 0  , 0  , 0  , 1   ] ] ;`
multmatrix(m)
    linear_extrude(height=10) 
        circle(r=5);
```

I found [this comment](http://forum.openscad.org/Multmatrix-and-its-mysterious-4th-row-for-idiots-td10506.html):

> nothing that I tried in the fourth row of the multmatrix function worked
>
>Most systems which are not interested in perspective calculations won't bother
to store the fourth row, since they "know" it will be [ 0 0 0 1 ].

Indeed I can change row 4 to [1,1,1,1] and nothing happens. So I wondered if at some point the 4x4 library was "optimised" to ignore row 4

This would therefore appear to be a bug either in operation, or in the documentation if MM does not actually do 4x4 transforms when it implies it does.

t-paul, 2020-04-12:
For some more notes on non-affine transformations, see #749.