openscad/openscad #3281: lambda generates syntax error

kamorrissey, 2020-04-14:
The trivial lambda example in the programming language reference does not compile in 2019.05 Windows release binary:

```
func = function (x) x + x;
echo(func(x));
```

Preview and render complain of a syntax error on line 1.

MichaelAtOz, 2020-04-14:
Function literals are in the development version, not released yet.
The wiki is deficient in not mentioning that, I'll fix it shortly
If you want to try a dev version you will also need to enable it in preferences/features.

MichaelAtOz, 2020-04-14:
Wiki updated. Closing.

closed 2020-04-14 by MichaelAtOz

kamorrissey, 2020-04-14:
So if I clone today's version and build, it will be there? I wanted to experiment with functional list iterating algorithms.

thehans, 2020-04-14:
Yes it will be in the current master branch.  You can also download a dev snapshot if you don't want to build it: http://www.openscad.org/downloads.html#snapshots