openscad/openscad #3286: make error: <QSaveFile>

AndreV84, 2020-04-17:
`/qt4/QtNetwork -I/usr/include/qt4/QtGui -I/usr/include/qt4/QtOpenGL -I/usr/include/qt4/QtMultimedia -I/usr/include/qt4 -Isrc -Isrc/ext/libtess2/Include -I/usr/include/qt4 -I/usr/X11R6/include -Iobjects -o objects/src/OpenSCADApp.o src/OpenSCADApp.cc

src/tabmanager.cc:4:10: fatal error: QSaveFile: No such file or directory
 **#include <QSaveFile>**
          ^~~~~~~~~~~
**compilation terminated.**

Makefile:3417: recipe for target 'objects/src/tabmanager.o' failed
make: *** [objects/src/tabmanager.o] Error 1
make: *** Waiting for unfinished jobs....
src/Preferences.cc: In member function ‘void Preferences::on_comboBoxLineWrapIndentationStyle_activated(int)’:
src/Preferences.cc:559:82: error: ‘class QComboBox’ has no member named ‘currentData’; did you mean ‘currentText’?
 rapIndentationIndent->setDisabled(comboBoxLineWrapIndentationStyle->currentData() == "Same" || comboBoxLineWrapIndentationStyle->currentData() == "Indented");
                                                                     ^~~~~~~~~~~
                                                                                  currentText
src/Preferences.cc:559:143: error: ‘class QComboBox’ has no member named ‘currentData’; did you mean ‘currentText’?
 Style->currentData() == "Same" || comboBoxLineWrapIndentationStyle->currentData() == "Indented");
                                                                     ^~~~~~~~~~~
                                                                                                                                               currentText
src/Preferences.cc: In member function ‘void Preferences::on_pushButtonOctoPrintCheckConnection_clicked()’:
src/Preferences.cc:723:85: error: no matching function for call to ‘QString::fromStdString(const QString)’
 ritical(this, _("Error"), QString::fromStdString(e.getErrorMessage()), QMessageBox::Ok);
                                                                     ^
In file included from /usr/include/qt4/QtCore/qobject.h:48:0,
                 from /usr/include/qt4/QtGui/qwidget.h:47,
                 from /usr/include/qt4/QtGui/qmainwindow.h:45,
                 from /usr/include/qt4/QtGui/QMainWindow:1,
                 from src/Preferences.h:3,
                 from src/Preferences.cc:27:
/usr/include/qt4/QtCore/qstring.h:1047:16: note: candidate: static QString QString::fromStdString(const string&)
 inline QString QString::fromStdString(const std::string &s)
                ^~~~~~~
/usr/include/qt4/QtCore/qstring.h:1047:16: note:   no known conversion for argument 1 from ‘const QString’ to ‘const string& {aka const std::__cxx11::basic_string<char>&}’
src/Preferences.cc: In member function ‘void Preferences::on_pushButtonOctoPrintSlicingEngine_clicked()’:
src/Preferences.cc:747:85: error: no matching function for call to ‘QString::fromStdString(const QString)’
 ritical(this, _("Error"), QString::fromStdString(e.getErrorMessage()), QMessageBox::Ok);
                                                                     ^
In file included from /usr/include/qt4/QtCore/qobject.h:48:0,
                 from /usr/include/qt4/QtGui/qwidget.h:47,
                 from /usr/include/qt4/QtGui/qmainwindow.h:45,
                 from /usr/include/qt4/QtGui/QMainWindow:1,
                 from src/Preferences.h:3,
                 from src/Preferences.cc:27:
/usr/include/qt4/QtCore/qstring.h:1047:16: note: candidate: static QString QString::fromStdString(const string&)
 inline QString QString::fromStdString(const std::string &s)
                ^~~~~~~
/usr/include/qt4/QtCore/qstring.h:1047:16: note:   no known conversion for argument 1 from ‘const QString’ to ‘const string& {aka const std::__cxx11::basic_string<char>&}’
src/Preferences.cc: In member function ‘void Preferences::on_pushButtonOctoPrintSlicingProfile_clicked()’:
src/Preferences.cc:783:85: error: no matching function for call to ‘QString::fromStdString(const QString)’
 ritical(this, _("Error"), QString::fromStdString(e.getErrorMessage()), QMessageBox::Ok);
                                                                     ^
In file included from /usr/include/qt4/QtCore/qobject.h:48:0,
                 from /usr/include/qt4/QtGui/qwidget.h:47,
                 from /usr/include/qt4/QtGui/qmainwindow.h:45,
                 from /usr/include/qt4/QtGui/QMainWindow:1,
                 from src/Preferences.h:3,
                 from src/Preferences.cc:27:
/usr/include/qt4/QtCore/qstring.h:1047:16: note: candidate: static QString QString::fromStdString(const string&)
 inline QString QString::fromStdString(const std::string &s)
                ^~~~~~~
/usr/include/qt4/QtCore/qstring.h:1047:16: note:   no known conversion for argument 1 from ‘const QString’ to ‘const string& {aka const std::__cxx11::basic_string<char>&}’
src/Preferences.cc: In member function ‘void Preferences::updateGUI()’:
src/Preferences.cc:944:88: error: ‘class QComboBox’ has no member named ‘currentData’; did you mean ‘currentText’?
 rapIndentationIndent->setDisabled(comboBoxLineWrapIndentationStyle->currentData() == "Same" || comboBoxLineWrapIndentationStyle->currentData() == "Indented");
                                                                     ^~~~~~~~~~~
                                                                                        currentText
src/Preferences.cc:944:149: error: ‘class QComboBox’ has no member named ‘currentData’; did you mean ‘currentText’?
 Style->currentData() == "Same" || comboBoxLineWrapIndentationStyle->currentData() == "Indented");
                                                                     ^~~~~~~~~~~
                                                                                                                                                     currentText
Makefile:2576: recipe for target 'objects/src/Preferences.o' failed
make: *** [objects/src/Preferences.o] Error 1
src/OpenSCADApp.cc: In member function ‘void OpenSCADApp::workaround_QTBUG_65592(QObject*, QEvent*)’:
src/OpenSCADApp.cc:28:51: warning: unused parameter ‘o’ [-Wunused-parameter]
 void OpenSCADApp::workaround_QTBUG_65592(QObject* o, QEvent* e)
                                                   ^
src/OpenSCADApp.cc:28:62: warning: unused parameter ‘e’ [-Wunused-parameter]
 void OpenSCADApp::workaround_QTBUG_65592(QObject* o, QEvent* e)
                                                              ^
src/openscad.cc: In function ‘int gui(std::vector<std::__cxx11::basic_string<char> >&, const boost::filesystem::path&, int, char**)’:
src/openscad.cc:709:8: warning: unused variable ‘s’ [-Wunused-variable]
  auto *s = Settings::Settings::inst();
        ^
`

t-paul, 2020-04-17:
Sorry, Qt4 is not supported anymore in `master`. Minimum Qt version is 5.4.

closed 2020-04-17 by t-paul

AndreV84, 2020-04-17:
yes, but qt5 is in the system
Moreover, checker finds qt5;
how to point out so that the program uses 5 version?

t-paul, 2020-04-17:
I don't know what your system is. On Debian based systems, there's a `qt5-default` package which should switch. Otherwise something like `QT_SELECT=5 qmake`. For other systems, please check their documentation.

AndreV84, 2020-04-17:
`qt5-default`
worked;
Thanks

AndreV84, 2020-04-17:
now it throws errors on
`/usr/include/GLES3/gl31.h:1502:146: error: ‘void __glewProgramUniformMatrix4x3fv(GLuint, GLint, GLsizei, GLboolean, const GLfloat*)’`
system aarch64 Ubuntu

AndreV84, 2020-04-17:
`/usr/include/GLES3/gl31.h:1518:88: error: ‘void __glewVertexBindingDivisor(GLuint, GLuint)’ redeclared as different kind of symbol
 GL_APICALL void GL_APIENTRY glVertexBindingDivisor (GLuint bindingindex, GLuint divisor);`

t-paul, 2020-04-17:
OpenSCAD requires OpenGL, it does not work with GLES. Most `aarch64` platforms support that in general, but it looks like it defaults to GLES on Ubuntu.