openscad/openscad #3290: Make Error: call to translation-make.sh not escaped/quoted

bpohl, 2020-04-20:
When the Makefile is generated it doesn't escape the fully qualified file path to call scripts/translation-make.sh.  It fails if there is a space in the path.  When I edit the Makefile to have either

  `/home/user/Git\ Repositories/openscad/scripts/translation-make.sh`

or

  `'/home/user/Git Repositories/openscad/scripts/translation-make.sh'`

it works fine.  

All other path references are escaped or quoted.  Seems that qmake misses doing so for translation-make.sh.

[And if anyone says `<mocking-voice>`"Just don't put spaces in dir/file names."`</mocking-voice>`, I'm going to kick them in the bytes!  Space is a VALID CHARTER! Don't be a lazy coder.]

donbright, 2020-04-21:
https://help.github.com/en/github/site-policy/github-community-guidelines

closed 2020-04-22 by t-paul

bpohl, 2020-04-29:
Thanks t-paul.

commit 253890e8 (2020-04-22): Extra quotes to fix build in directories with spaces (fixes #3290).
commit de778db7 (2020-04-22): Merge pull request #3292 from openscad/fix-spaces-in-path

Extra quotes to fix build in directories with spaces (fixes #3290).