openscad/openscad #3291: Error on building for Raspberry Pi 3

NervenCid, 2020-04-21:
Hello:

I'm following this tutorial in order to build openscad for RaspBerry Pi 3:

https://github.com/koendv/openscad-raspberrypi

But i'm getting these errors:

```
src/Preferences.cc: In member function ‘void Preferences::on_comboBoxLineWrapIndentationStyle_activated(int)’:
src/Preferences.cc:559:82: error: ‘class QComboBox’ has no member named ‘currentData’; did you mean ‘currentText’?
 rapIndentationIndent->setDisabled(comboBoxLineWrapIndentationStyle->currentData() == "Same" || comboBoxLineWrapIndentationStyle->currentData() == "Indented");
                                                                     ^~~~~~~~~~~
                                                                     currentText
src/Preferences.cc:559:143: error: ‘class QComboBox’ has no member named ‘currentData’; did you mean ‘currentText’?
 Style->currentData() == "Same" || comboBoxLineWrapIndentationStyle->currentData() == "Indented");
                                                                     ^~~~~~~~~~~
                                                                     currentText
src/Preferences.cc: In member function ‘void Preferences::on_pushButtonOctoPrintCheckConnection_clicked()’:
src/Preferences.cc:723:85: error: no matching function for call to ‘QString::fromStdString(const QString)’
 ritical(this, _("Error"), QString::fromStdString(e.getErrorMessage()), QMessageBox::Ok);
                                                                     ^

In file included from /usr/include/qt4/QtCore/qobject.h:48,
                 from /usr/include/qt4/QtGui/qwidget.h:47,
                 from /usr/include/qt4/QtGui/qmainwindow.h:45,
                 from /usr/include/qt4/QtGui/QMainWindow:1,
                 from src/Preferences.h:3,
                 from src/Preferences.cc:27:
/usr/include/qt4/QtCore/qstring.h:1047:16: note: candidate: ‘static QString QString::fromStdString(const string&)’
 inline QString QString::fromStdString(const std::string &s)
                ^~~~~~~
/usr/include/qt4/QtCore/qstring.h:1047:16: note:   no known conversion for argument 1 from ‘const QString’ to ‘const string&’ {aka ‘const std::__cxx11::basic_string<char>&’}
src/Preferences.cc: In member function ‘void Preferences::on_pushButtonOctoPrintSlicingEngine_clicked()’:
src/Preferences.cc:747:85: error: no matching function for call to ‘QString::fromStdString(const QString)’
 ritical(this, _("Error"), QString::fromStdString(e.getErrorMessage()), QMessageBox::Ok);
                                                                     ^

In file included from /usr/include/qt4/QtCore/qobject.h:48,
                 from /usr/include/qt4/QtGui/qwidget.h:47,
                 from /usr/include/qt4/QtGui/qmainwindow.h:45,
                 from /usr/include/qt4/QtGui/QMainWindow:1,
                 from src/Preferences.h:3,
                 from src/Preferences.cc:27:
/usr/include/qt4/QtCore/qstring.h:1047:16: note: candidate: ‘static QString QString::fromStdString(const string&)’
 inline QString QString::fromStdString(const std::string &s)
                ^~~~~~~
/usr/include/qt4/QtCore/qstring.h:1047:16: note:   no known conversion for argument 1 from ‘const QString’ to ‘const string&’ {aka ‘const std::__cxx11::basic_string<char>&’}
src/Preferences.cc: In member function ‘void Preferences::on_pushButtonOctoPrintSlicingProfile_clicked()’:
src/Preferences.cc:783:85: error: no matching function for call to ‘QString::fromStdString(const QString)’
 ritical(this, _("Error"), QString::fromStdString(e.getErrorMessage()), QMessageBox::Ok);
                                                                     ^

In file included from /usr/include/qt4/QtCore/qobject.h:48,
                 from /usr/include/qt4/QtGui/qwidget.h:47,
                 from /usr/include/qt4/QtGui/qmainwindow.h:45,
                 from /usr/include/qt4/QtGui/QMainWindow:1,
                 from src/Preferences.h:3,
                 from src/Preferences.cc:27:
/usr/include/qt4/QtCore/qstring.h:1047:16: note: candidate: ‘static QString QString::fromStdString(const string&)’
 inline QString QString::fromStdString(const std::string &s)
                ^~~~~~~
/usr/include/qt4/QtCore/qstring.h:1047:16: note:   no known conversion for argument 1 from ‘const QString’ to ‘const string&’ {aka ‘const std::__cxx11::basic_string<char>&’}
src/Preferences.cc: In member function ‘void Preferences::updateGUI()’:
src/Preferences.cc:944:88: error: ‘class QComboBox’ has no member named ‘currentData’; did you mean ‘currentText’?
 rapIndentationIndent->setDisabled(comboBoxLineWrapIndentationStyle->currentData() == "Same" || comboBoxLineWrapIndentationStyle->currentData() == "Indented");
                                                                     ^~~~~~~~~~~
                                                                     currentText
src/Preferences.cc:944:149: error: ‘class QComboBox’ has no member named ‘currentData’; did you mean ‘currentText’?
 Style->currentData() == "Same" || comboBoxLineWrapIndentationStyle->currentData() == "Indented");
                                                                     ^~~~~~~~~~~
                                                                     currentText
make: *** [Makefile:2576: objects/src/Preferences.o] Error 1
```

I don't have very deep knowledge

Thank you and sorry for the english

Scopeuk, 2020-04-21:
The above log shows that QT version 4 was used for the build, as a minimum openscad requires QT version 5.2 please refer to your distributions documentation to install build dependencies.
a list of dependencies can be found at
https://github.com/openscad/openscad#prerequisites

t-paul, 2020-04-21:
In addition to Qt5, also the OpenGL backend is required. That is possible on Raspberry Pi, but most precompiled versions use GLES which is not compatible with OpenSCAD.

closed 2020-04-21 by t-paul

t-paul, 2020-06-15:
OpenSCAD builds fine and works on Raspberry PI OS 64-bit. This is still in beta, but should hopefully appear on the official download page soon.