openscad/openscad #3293: 2019.12.20 Mac OSX Preview does not complete on part, but Render does complete

valerioa, 2020-04-24:
I have a part that has a cylinder with a top edge bevel. I use obijuan's obiscad library to do beveling ( https://github.com/Obijuan/obiscad ) . So far obiscad beveling has always worked for me. 
This is the part (beveling colored in red)

[image]

I use a global_fn parameter that drives $fn parameters for the cylinder and the top edge bevel segmentation.

With global_fn up to 38, everything works as expected, both Preview and Render can render the part.

With global_fn = 39, Preview takes a long time. It might render, but, in a time longer than my patience can take, I did not wait for it.

With global_fn >= 40, the Preview does not render the part, but, Render does:

[image]

Not sure $fn larger than 30 are useful for roundness when parts are 3d printed, but, I thought I'd report it anyways:

Here's the source code:

```
global_fn = 40;

center_leg_center = [0.0, 0.0];

cylinder_body_height = 20.0;
cylinder_tip_height  = 10.0;

cylinder_body_diameter = 17.0;
cylinder_tip_diameter  = 10.0;

use <bevel.scad>;

function mod(v) = (sqrt(v[0]*v[0]+v[1]*v[1]+v[2]*v[2]));
function unitv(v) = v/mod(v);

//-- Generate bevel connectors
//-- p1 point (x,y,z triplet) at the start of the edge
//-- p2 point (x,y,z triplet) at the end of the edge
//-- ft1, a triangle ( a vector of three x,y,z triplets ), coplanar to one of the two faces separated by the edge
//-- ft2, a triangle ( a vector of three x,y,z triplets ), coplanar to the other of the two faces separated by the edge
//-- triangle points must be ordered according to the right hand rule

function getBevelConnectors(p1, p2, ft1, ft2) =
	[
		[
			((p1+p2)/2),         // point
			(p2-p1)/norm(p2-p1), // vector
			0                    // rotation
			],
		[
			((p1+p2)/2),                                                // point
			unitv( (unitv(cross( ft1[2] - ft1[1], ft1[0] - ft1[1])) +
			        unitv(cross( ft2[2] - ft2[1], ft2[0] - ft2[1]) ))), // vector
			0                                                           // rotation
			]
		];

// c = center of top cylinder face
// d = cylinder diamter
// br = bevel diameter
module top_cylinder_bevel(c, d, br) {
     for(a=[0:(360/global_fn):360]) {
	  bconn = getBevelConnectors(
	       [c[0]+cos(a)*d/2, c[1]+sin(a)*d/2, c[2]],
	       [c[0]+cos(a+360/global_fn)*d/2, c[1]+sin(a+360/global_fn)*d/2, c[2]],
	       [  [c[0]+cos(a)*d/2, c[1]+sin(a)*d/2, c[2]], [c[0]+cos(a+360/global_fn)*d/2, c[1]+sin(a+360/global_fn)*d/2, c[2]], c] ,
	       [  [c[0]+cos(a)*d/2, c[1]+sin(a)*d/2, c[2]], [c[0]+cos(a)*d/2, c[1]+sin(a)*d/2, c[2]-10], [c[0]+cos(a+360/global_fn)*d/2, c[1]+sin(a+360/global_fn)*d/2, c[2]]]
	       );
	  color ("Red")
	       bevel(
		    bconn[0],
		    bconn[1],
		    cr = br ,
		    cres=6,
		    l=norm([c[0]+cos(a+360/global_fn)*d/2, c[1]+sin(a+360/global_fn)*d/2, c[2]] -
			   [c[0]+cos(a)*d/2, c[1]+sin(a)*d/2, c[2]])+1);
     }
}

module center_pillar() {
     union() {
          cylinder(h=cylinder_body_height, d=cylinder_body_diameter, $fn= global_fn);
          difference() {
               translate([0, 0, cylinder_body_height])
                    cylinder(h=cylinder_tip_height, d=cylinder_tip_diameter, $fn= global_fn);
               top_cylinder_bevel([0, 0, cylinder_body_height+cylinder_tip_height], cylinder_tip_diameter, br=1.5);
          }
     }
}

center_pillar();
```
Attached a zip file containing all source files needed for rendering.

[bug_preview.zip](https://github.com/openscad/openscad/files/4530323/bug_preview.zip)

t-paul, 2020-04-24:
As the warning says, you hit a limit in processing. You can try changing that value in Preferences->Advanced, higher values may mean the re-draw times can get so high that depending on the system and graphics driver, the system gets unresponsive.
Alternatively force mesh calculation via `color("red") render() bevel(...)`. This will impact preview performance the first time though.

nophead, 2020-04-24:
Seems like a complicated and slow way to make that shape. I would just
rotate_extrude a rounded corner unioned with a square. That would render
and preview instantaneously even with large $fn. Rounded corner is here:
https://github.com/nophead/NopSCADlib#Rounded_cylinder

valerioa, 2020-04-24:
@t-paul render() works for me thanks for that.

@nophead : for this simple shape yes. For a general case, where the edge form is arbitrary and the edge orientation in 3d space is arbitrary, obiscad's bevel solves the general case.

closed 2020-04-24 by t-paul