openscad/openscad #3295: Access denied whilst trying to save in OpenSCAD

jmcbassett, 2020-04-26:
I've had a strange problem for a while now, which I've finally determined must be an issue with OpenSCAD.

When I try to save (overwrite) a modified (using OpenSCAD) file to my network drive (NAS) I always get an "Error saving design" pop-up, which also says "Access is denied.)".  I don't get the error when using "save as" to save to a new file (on the network drive), but will using "save as" to try overwriting a file.  I also don't get the error when overwriting an existing file with OpenSCAD if it's on one of my PCs internal drives.  OpenSCAD is happy to overwrite STL files but not .scad

If I open the OpenSCAD file in Notepad, Windows (10 Pro 64-bit) is happy to overwrite the existing file.  I also never get the error with any other file type or program.

I only ever get the message using OpenSCAD to overwrite an existing file that is on my NAS.

I'm at a loss as it what is causing the error.

MichaelAtOz, 2020-04-26:
I just did all that from my W7/64 to (mapped to B:) ReadyNAS NAS, no problem.

jmcbassett, 2020-04-26:
> I just did all that from my W7/64 to (mapped to B:) ReadyNAS NAS, no problem.

Okay thanks, but that doesn't explain why OpenSCAD doesn't work correctly on my PC with Windows 10 Pro and a different NAS.

nophead, 2020-04-26:
It looks like QSaveFile is used to save the file. https://doc.qt.io/qt-5/qsavefile.html

It writes a temporary and then moves it to the final file. That might explain why other programs don't have the same problem.

t-paul, 2020-04-26:
Similar issue in #3071.

jmcbassett, 2020-04-26:
> Similar issue in #3078.

Not that similar in my opinion as I just get a pop error.  I have seen no evidence of the content of the original file being changed at all, just a refusal to change it; certainly no evidence of the "NUL" issue.

jmcbassett, 2020-04-26:
> It looks like QSaveFile is used to save the file. https://doc.qt.io/qt-5/qsavefile.html
> 
> It writes a temporary and then moves it to the final file. That might explain why other programs don't have the same problem.

I'm not familiar with QSaveFile, but OpenSCAD happily writes a new file using "Save as".  I suppose it could write the new file and fail to remove the original, but I've seen no 'new' files appearing from a failed save..

nophead, 2020-04-26:
Does move work on your NAS? I have had NASs with bugs in the past.

It could also be a timing issue. I have had to insert 0.1s delays in some
of my Python scrips that deleted a directory and then recreated it. Perhaps
a delete followed by a move immediately afterwards causes an error on the
NAS. Or perhaps move immediately after closing a write.

jmcbassett, 2020-04-27:
> Does move work on your NAS? I have had NASs with bugs in the past. It could also be a timing issue. I have had to insert 0.1s delays in some of my Python scrips that deleted a directory and then [...]

Yes, move works, the only thing that doesn't is overwriting an existing file in OpenSCAD.

nophead, 2020-04-27:
I don't think it ever actually overwrites though. It writes to a temporary and then it must delete the old file and rename the temporary. 

Can you make a batch file that does that to see if your NAS gives a permission error when you do those in quick succession?

Since OpenSCAD delegates to saving to QT I think it is either a NAS or QT issue rather than OpenSCAD.

jmcbassett, 2020-04-27:
The batch file ran without an error.

The issue could be with QT but I don't have any knowledge of that tool, nor any way to rectify the problem.  I use OpenSCAD as a designer, I'm not a software engineer, and although I have some knowledge, it's limited.  Presumably QT is coded in to OpenSCAD so it would need tweaking to be done to the OpenSCAD code?

nophead, 2020-04-27:
If it is bug in QT then the problem would need to be posted upstream and wait for QT developers to fix it. However if it is specific to one make of NAS it would be for the NAS developers to fix or at least report to QT what is wrong.

All OpenSCAD does is use QSaveFile which is coded to handle detecting full devices rather than overwriting with a zero length file that a naive home rolled version would do. The code is here: https://github.com/openscad/openscad/blob/9c1070820bd91aaf9589871f6aa07f2ac8d42ae6/src/tabmanager.cc#L654

Saving over a file works fine on my Seagate NAS, so I don't know how it can be debugged unless a developer has your make of NAS.

The QT code is here: https://code.woboq.org/qt5/qtbase/src/corelib/io/qsavefile.cpp.html#_ZN9QSaveFile6commitEv

jmcbassett, 2020-04-27:
Thanks but unfortunately the code is about as much use as a chocolate tea pot to me.

Strangely the problem doesn't occur with .stl files, which presumably use the same code?

nophead, 2020-04-27:
No they don't, presumably because it doesn't matter if you lose an STL when your disk is full but it isn't good to lose a source file.

jmcbassett, 2020-04-27:
> No they don't, presumably because it doesn't matter if you lose an STL when your disk is full but it isn't good to lose a source file.

Interesting as saving .stl files works and saving .scad doesn't

nophead, 2020-04-27:
Saving an STL is a simple write to a file using C++ standard library and doesn't use QT because it has to work in command line mode. If something goes wrong during that write you would end up with a duff stl, perhaps zero length if your disk is full.

When saving the source code from the GUI QT is available and so QSaveFile is used to make sure it always leaves a valid file, even if the write fails. So it does the more complicated thing of writing to a temporary and checking that worked and the renaming it as the last step, so as you have found the original file is preserved if it fails.

It is hard to see why it fails on your NAS but not on two others and not for millions of users of QT. My best guess is there is a bug in your NAS.

Do you have any of the programs on this list of programs that use QT https://en.wikipedia.org/wiki/Category:Software_that_uses_Qt? If so you could see if Save works in those on your NAS.

Another thing you could try is removing the source file before you save. I expect that would work if SaveAs does. It seems likely that the rename goes wrong.

jmcbassett, 2020-04-28:
The only program on the list I had installed was Google Earth Pro and had no issues accessing and saving files to the same folder on the NAS; however, I'm not sure that still uses QT.

To try and aid diagnosis I downloaded MakeHuman and LibreCAD that are on the list and both were happy to save and overwrite files on that NAS folder.  Maybe QT isn't the cause of the issue?

nophead, 2020-04-28:
OpenSCAD only prints "Error saving design"  when ```QTextStream.status()``` doesn't return ```QTextStream::Ok``` after writing the file or ```QSaveFile.commit()``` returns false. The "Access is denied" string comes from QSaveFile.errorString().  So whatever goes wrong must be inside QT.

We also know that opening the file to write succeeds. 

I don't think you can get a permission error during a write if you successfully opened it for writing, so I think most likely ```QSaveFile.commit()``` is failing. That is the part that closes the file and renames it. On Windows it ends up here: https://code.woboq.org/qt5/qtbase/src/plugins/platforms/winrt/qwinrtfileengine.cpp.html#385, which calls ```RenameAsync(nativeDestinationName.Get(), NameCollisionOption_ReplaceExisting, &op);``` which seems to rename the temporary file to replace the existing file. To be honest I didn't know that was possible without deleting the original first. Perhaps it isn't implemented on your NAS.

jmcbassett, 2020-04-28:
Is there anyway to simply test that function on my NAS?

nophead, 2020-04-28:
There is a move command which asks if it should overwrite. You could try that but I don't know if it actually uses NameCollisionOption_ReplaceExisting or just removes the original first. Obviously if QSaveFile removed the file first you could lose your source file, which is why it uses the special option.

jmcbassett, 2020-04-28:
A move command attempting to overwrite, prompted confirmation and when given, resulted in an "Access is Denied" error message.

So we have some form of result; I just need to sea if there are any settings in the Seagate NAS control panel to change this behaviour.

jmcbassett, 2020-04-28:
I'd already checked and there is no firmware update available; as usual manufacturers lose interest very quickly once a model is superseded.

I swap between my desktop and laptop so use the NAS as my file store for anything that might need access from either machine.

Might be time for a replacement storage system

nophead, 2020-04-28:
I use GoogleDrive and a NextCloud Pi server for sharing stuff between machines.

My Seagate firmware is 2015.0916.0008-F from 2015 and says it is the latest.

jmcbassett, 2020-04-28:
My NAS is firmware v2015.00330 but is probably an earlier version as no update is available.

How easy is it to set-up a NextCloud Pi server and does it support RAID?

nophead, 2020-04-28:
It was very easy to set up on a RPI4 with a 1TB USB SSD. I access it only via a VPN when away from home to reduce the security risk. 

I don't know if it does RAID. It currently backs up itself onto another partition of the same SSD. As it is a copy of what is on my primary machine and that is backed up to the NAS I don't consider it vital. It is mainly so I can work on my projects from my winter residence in Tenerife.

jmcbassett, 2020-04-28:
My NAS is the master and has mirror (type 1) RAID to protect the data.  I do periodically backup to my desktop and back that up to an external drive.

The NAS is not available from the internet, only from my local Wi-Fi, but it allows me to share between my main desktop, a mini-PC for my CNC mill, my laptops (3D printing) and even my iPad.

As I don't go away for long periods I can usually download what I need before going; remote access would be an advantage but I never got it working on my Seagate NAS.

nophead, 2020-04-28:
Mine was supposed to do remote access but I didn't think it was a good idea to put my entire data online via a third party. I got an email three year ago to say it was no longer available.  Great, you buy a product but a major feature of it just disappears. Good job I wasn't using it.

Nowadays software stops working long before the hardware breaks.

jmcbassett, 2020-04-28:
I got the same email, but had never succeeded in getting it to work anyway.

One of the big problems is that hardware designs move on very quickly and support of the firmware & software often declines rapidly once the design is a couple of generations old; as little as a year in some cases.

MichaelAtOz, 2020-04-28:
With ReadyNAS there are regular firmware updates, you also have the option to pay for support, I'm not saying they're perfect, but I'd buy another when this one dies. I also use SpiderOak on my source, changes promptly copied to a very secure cloud, shareable on various devices.

jmcbassett, 2020-04-29:
> With ReadyNAS there are regular firmware updates, you also have the option to pay for support, I'm not saying they're perfect, but I'd buy another when this one dies. I also use SpiderOak on my source, changes promptly copied to a very secure cloud, shareable on various devices.

Thanks, but for now I can live with what I've got; it's just a bit irritating having to "save as" and manually remove unwanted files.

closed 2020-04-29 by jmcbassett