openscad/openscad #3296: Multi-tab version should have multiple image buffers

nophead, 2020-04-26:
The new multi-tab editor doesn't make sense when you switch back to a previous tab because it doesn't draw the view that that tab generated. One has to hit F5 to make the view match the tab each time you switch tabs. That is tedious and can be time consuming if the preview takes significant time.

It would be much better if the view was saved in an off screen buffer and restored. If that is not possible at least do an auto F5 when switching tabs like it does when opening a new one.

t-paul, 2020-04-26:
That's by intention and as there's no means of a "project" which combines multiple files/tabs it's not going to catch all use cases.

nophead, 2020-04-26:
I don't understand. Whenever a new tab is opened it is automatically previewed and draws what the file produces, often a blank screen in my case if it is a library file but if it is a project file it will draw an assembly view. If I select a previous tab I expect it to show the view corresponding to that, as it it did when first opened. I don't see how it is related to use case. The view should simply correspond to whatever file is in the currently selected tab.

MichaelAtOz, 2020-04-26:
I didn't see tabs being very useful with only one preview.

But there are use cases where you have your main source file, previewed, see a problem, then can click a library file tab, while still viewing the problem. Most of my libraries don't create objects, so previewing on a tab switch would be unhelpful. 

So it will probably need some options. I wouldn't like previews firing off on tab changes.

One option would be to save the framebuffer WHEN doing F5, so when tabbing back to a previously previewed tab it paints the screen, rather than doing a compile/preview. Tabbing back to a library NOT previewed leaves whatever image as is.

MichaelAtOz, 2020-04-26:
May also have different behaviour with Autoload&preview, perhaps have that set per tab?

nophead, 2020-04-26:
The issue really comes to light with #3283. You click on an object on the screen and it takes you to the file that created that element and then previews it, erasing the main image. You flip back to your original tab and expect to see the view you came from so you can click on another part but you have to hit F5, which takes an age for a bi project.

If it could save the framebuffer for each tab it becomes far less clunky and you probably don't need per tab options to auto preview. Or perhaps a simpler solution is to not auto preview a new tab so the original image is not removed.

I don't normally use the internal editor so I always have auto preview on. Perhaps simply turning it off make more sense.

nophead, 2020-04-27:
I think if auto preview is off then the current behaviour makes sense. You just have one view and you can manual preview which tabs you want.

If auto preview is on it should always show the tab selected. An improvement would be to preview on tab switch but much better would be to save and restore frame buffers.

MichaelAtOz, 2020-04-27:
Is the frame buffer enough, does the 'preview state' ie ability to rotate/zoom still exist?

nophead, 2020-04-27:
Good point, no there is currently only one camera state, so that would need to be stored per tab as well.

Tomatower, 2020-04-27:
another idea is to make the preview tabbed as well, but independent of the editor tabs.
if i have a main.scad that draws the main model, i dont want to tab back-and-forth to adjust one feature deep down in the file hierarchy and get another preview of the main model.

For example:
Open a file / press F5 you get a tab on the render preview named the same as your file.
You have a checkbox between the windows to bind tab switches together (default: on)

Benefits:

* you get auto-preview / last rendered entity for the currently selected tab. (ideally for having several small models laying around, simple projects)
* you get to edit a component in the general where your main.scad is updated and re-rendered when you save a file within its tree.

So I'd suggest to create several QGLViews within a tabbed editor (and yes - wasting a huge amount of ram...)