openscad/openscad #3297: Include and use file underscore links get mixed up

nophead, 2020-04-26:
[image]

Some random lines get underscores and link to random files in the project. Some of the correctly underscored files link to the wrong file.

Latest master: OpenSCAD 2020.04.26.ci4968

t-paul, 2020-04-26:
Do you have a smaller example. I'm currenty trying to reproduce that and so far I failed. I assumed it happens with same file names in different folders, but:

debug.scad
x/debug2.scad
x/debug3.scad

with debug including both debug2 and debug3 and debug2 including debug3 it does not seem to trigger.

t-paul, 2020-04-26:
Actually, maybe small is not so relevant, I'll try with the libtest.scad which seems to cause some of the misplaced markers too.

nophead, 2020-04-27:
Yes it's weird. For example washer.scad gets a link to math.scad, which it doesn't use but it does use sweep.scad that uses maths.scad.

nophead, 2020-04-27:
I think it might simply be because nested include files are added to the FileModule includes but are not actually in that file. they are actually in the file it includes.

nophead, 2020-04-27:
Similarly I think ```use```s that are in an included file get attributed to the FileModule was well. I.e. it represents a flat file produced by expanding all the includes. So they all get markers but really it is only the direct includes and uses that should have markers in the FileModule itself.

t-paul, 2020-05-09:
Yep, exactly that, the parser did not know if it was parsing the main file or an include.

commit 4f4981a6 (2020-05-22): Prevent include files to register links (fixes #3297).
commit cab39bba (2020-05-22): Merge pull request #3306 from openscad/fix-include-links

Prevent include files to register links (fixes #3297).
closed 2020-05-22 by t-paul