openscad/openscad #3298: Using rotate_extruded torus in difference causes assert

camylarde, 2020-05-01:
This code (simplified) produces assert error when rendering the object
`difference(){
	translate([-2.5,-2.5,0])cube([5,5,5]);
rotate_extrude(convexity=10)translate([5,0,0])circle(r=5);}`

> ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation! Expr: e->incident_sface() != SFace_const_handle() File: /mxe/usr/x86_64-w64-mingw32.static.posix/include/CGAL/Nef_S2/SM_const_decorator.h Line: 329

The object is shown normally without an assert when using preview though.

nophead, 2020-05-01:
You have a point of singularity in the centre of your torus. Translate it by slightly more than 5 to give a finite sized hole in the middle and it will then be a manifold required by CGAL.

TLC123, 2020-05-01:
Does rotate_extruded need to be nurfed  by always auto clipping everything  x =< 0 ?
Nah wouldn't fix single vertex on center line anyways.

nophead, 2020-05-01:
No you can have things on the centreline in general. For example rotating a semicircle produces a sphere. What you can't have is a torus with a singularity in the centre where the two opposite faces come together and touch each other. It is no longer topologically a torus or a sphere.

If you translate the circle positively in x you get a torus with a tiny hole through the middle. If you translate it negatively and clip it to the y axis you a topological sphere with dimples in it. Both are valid 2-manifold surfaces and are physically possible. What you have made is physically impossible because it has a point of zero thickness in the centre.

thehans, 2020-05-23:
One idea to help with this would be to have rotate_extrude detect if the polygon has a single point at x==0, and if so then silently insert a second point with miniscule vertical offset to give it *some* thickness there.  
Would just need to make sure to offset in the correct direction and insert before/after correspondingly to make sure the modified polygon doesn't self intersect.

nophead, 2020-05-24:
Seems ambiguous to me. You could offset in either direction and by how much? Also it could be more than one single point. Any point on the y axis that isn't one end of an edge parallel to the axis.

I prefer OpenSCAD to leave vertices where I program them to be. Not fudge around with them.

thehans, 2020-05-24:
Good point about the possibility of multiple points on the axis. I hadn't considered that, but the same idea could still be applied to multiple points.

The amount would be whatever minimum is determined enough for CGAL to not complain.  
I would probably try the next representable floating point value using: https://en.cppreference.com/w/cpp/numeric/math/nextafter

I'd have to double check the code, but think most of the polyset is actually using double precision, and only converting to single precision in a couple cases such as exporting to STL.  If double precision delta is possible it would be on the otrder of a few *attometers* (assuming the model is using mm for units).  It may require some testing to determine if that is sufficient to satisfy CGAL though.  
Even if this must be done using single precision floating point, if I've done my math right, such a delta should be on the order of a nanometer or so.

As far as whether to put it below or above the existing point, the idea I had in mind which I didn't really get into details before: 
1) For each "lone point" on the Y-axis, look at its two neighboring vertices.  
2) If one of those neighbors has the same y value as the original point, then insert the new point in whichever direction the *other* vertex is offset.  This ensures that flat tops/bottoms stay flat.  
If all three are on the same y axis coordinate then throw an error since that's degenerate zero area triangle anyways.  
If neither neighbors have the the same y value as the original, then maybe split the difference and move the original point slightly one way and have the inserted point go slightly the other way.

nophead, 2020-05-24:
It might have to be the dreaded grid's next value to avoid being broken by PolySet.

kintel, 2024-03-31:
Status as of 2024-03-24: Still broken with CGAL, but works with Manifold