openscad/openscad #3299: Feature request: additional include("filename") syntax

nalimcos, 2020-05-03:
Hello!

First of all thanks to everyone working on this project; it has been extremely useful to me over the past few years.

**TLDR version**
- Allowing `include("filename");` syntax would be useful in complex projects sharing parameter values with other programming languages.
- If this is seen as a useful addition, I may be able to write a patch.

**A bit of context** (skip if uninterested in integrating multiple languages in a single project)

I have been using OpenSCAD as part of a complex project (a fully customizable saxophone mouthpiece), where I need a few dozen parameters to be shared with another programming language used to generate data files then used with surface().

First moved them to a separate file, included both from SCAD and C code - chose C due to somewhat similar syntax.
- works but relies on undefined/old-standard behaviour
- undeclared variables assumed by GCC to be integers; original variables need to be multiplied by some large number to have any sort of precision, and re-divided later on, i.e. a bit of a mess.

Then tried python, and finally discovered JavaScript was closer in syntax to OpenSCAD and ported my code to node.js. (side benefits: no more extra compilation step -> less files floating around + full-blown makefile replaced with tiny convenience scripts -> much less code, much shorter commands when creating new prototype, saves a LOT of time)
- shared include file **works perfectly** (in non-strict mode, undeclared but defined variables are allowed; just watch out for overrides)
- just had to add a small compatibility layer: `require('compat.js')` with stuff like `PI=Math.PI;
tan=Math.tan;`

**Conditional includes** - may be nice to have but the more I think about it, the less it seems necessary.

In my project, I need a large number of variables (currently 24) to be set depending on a single `type` string which will have 5 possible values (plus several more during development)
- While I believe ternary operators could work, they would be extremely cumbersome to work with, especially when adding a new type or making a small modification to an existing `type` preset
- Arrays of variables and a type index may work, at the expense of readability - also, unsure about syntax compatibility.
- Obviously, I could just define `type` in the included files and modify the include statement; however, suboptimal w.r.t. next section.
- Some sort of conditional include would be nice; maybe this could be achieved by supplying a filename string 'variable' to include, rather than necessarily specifying the filename on the same line? But then again, I guess include statements are processed before parsing anything else, so this would be way messier than it looks?

**Interoperability** - here is the important part

This request is more specific to the multi-programming-language project case.

Current include syntax is `include <filename>;`
I would like to suggest adding the following variation in order to improve interoperability with other languages:
- `include("filename");` would avoid syntax errors when sharing file with e.g. node.js / python / ... , allowing one to define a compatibility layer function to handle said include. (currently, I use a hack: include lines are filtered out and processed by a bash script, converted to a require() path then fed as a command-line argument, while the parent shared include and main script are fed to nodejs stdin via `grep -v include`... it's a mess, but works within a few lines of code)
- Similarly, `#include "filename";` and allowing variable declarations to be prefixed with e.g. `double` would help when working with C/C++/..., but this is maybe too much. I get the impression macros would quickly get very messy in this case - one might just write a three-line conversion script and call it a day.

t-paul, 2020-05-03:
Why is there a need to share actual code snippets and not use data files? E.g. prototype PR #3087 has a JSON import as example.

nalimcos, 2020-05-03:
Thanks, that looks interesting. I have some (simple, valid openscad/c/javascript/python) math going on in there though, will that still work when moved to a JSON file?

t-paul, 2020-05-03:
No, I don't think JSON supports expressions. Those would need to go into the code loading the data. Or maybe there's a format that would support embedded expressions.

toddInPortland, 2020-07-04:
To extend the JSON suggestion, I think you would get portability and reusability to declare all your varibles in a separate JSON file, and as a separate step use python to read the JSON file and then generate an scad file and a c file. This way you can avoid conflicts between scad and c, as well as crafting each output file to take advantage of the aspects of each language.  For example, you could take a collection of variables in the JSON file and generate enums in C, making it part of a strong typing aspect of C, and then generate explicit asignments in scad.